userx14/omblepy #2: Documentation

bonnybabukachappilly, 2022-07-11:
Can you tell me how you find which UUIDs for different functions and bytes code to write for a particular action?

userx14, 2022-07-11:
Ok I have just seen that you are using omron 7361t

The reason that this one is not supported yet is, that is has a different byte order in the records itself. If you can give me some time I can hopefully work it out. If you want to do some testing in the meantime you should be able to use:
```
import sys
import datetime
sys.path.append('..')
from utility import bytearrayBitsToIntLittleEndian

def recordSize():
    return 0x10
def startAddress():
    return 0x98
def user1Entries():
    return 100
def user2Entries():
    return 100

def recordToDict(recordBytes):
    recordDict = dict()
    print(f"{recordBytes}")
    #Well that's the reason I have no added support yet
    recordDict["datetime"] = datetime.datetime(year, month, day, hour, minute, second)
    return recordDict
```

Maybe you could give me one or two records bytesArrays to test on.

bonnybabukachappilly, 2022-07-11:
Let me try, I will get back to you.

bonnybabukachappilly, 2022-07-11:
'The reason that this one is not supported yet is, that is has a different byte order in the records itself'

what records

userx14, 2022-07-11:
> 'The reason that this one is not supported yet is, that is has a different byte order in the records itself'

So each measurement you take is stored in the eeprom with the time of recording, the sys pressure, the dia pressure, and the heart rate and some other stuff. 
Omron calls each such stored measurement a "record", so I have copied their nomenclature here.
Presumabelly to save eeprom space and cut down the transmission time, Omron has decided to compress the records were they can. For my device this results in:
[image]
And that are the values you see hardcoded in https://github.com/userx14/omblepy/blob/63fd4919bfb851d23e1bf031cbc054eb9627e2ed/deviceSpecific/hem-7322t-d.py#L17-L27
And in the code of UBMP this corresponds to:
https://codeberg.org/LazyT/ubpm/src/commit/815a74dce1c41b16594e9e2d6208f96f3f2999a2/sources/plugins/vendor/omron/hem-7361t/DialogImport.cpp#L442-L448

And to complicate things further your device seems to use a different endianess.

LazyT, 2022-07-11:
0x820F happens for me, if the device isn't paired in OS.

bonnybabukachappilly, 2022-07-11:
When connecting to the device after paring, windows will give a prompt to pair.
if paired or not will get 'Same transmission failed 5 times, abort'
when trying 'again bleak.exc.BleakError: Could not get GATT services: Unreachable'
so I need to remove the device from windows, and the same when repeating

bonnybabukachappilly, 2022-07-11:
if i give 
await bleClient.write_gatt_char(self.deviceUnlock_UUID, b'\x01' + b'\x00'*16, response=True) insted of 
await bleClient.write_gatt_char(self.deviceUnlock_UUID, b'\x02' + b'\x00'*16, response=True) 

I get a 0x8104

userx14, 2022-07-11:
> When connecting to the device after paring, windows will give a prompt to pair. if paired or not will get 'Same transmission failed 5 times, abort' 

Does the initial pairing with -p work for you? Does the programm print something like "Paired device successfully with new key..."?

> when trying 'again bleak.exc.BleakError: Could not get GATT services: Unreachable' so I need to remove the device from windows, and the same when repeating

I also have this issue, sometimes removing the device from the windows bluetooth settings seems to be neccesary

bonnybabukachappilly, 2022-07-11:
> 

I can't say paring work, there is 0x01 and 0x02 available with program mode, if 0x02 get a response of 0x820F and for 0x01 0x8104

userx14, 2022-07-11:
> if i give await bleClient.write_gatt_char(self.deviceUnlock_UUID, b'\x01' + b'\x00'*16, response=True) insted of await bleClient.write_gatt_char(self.deviceUnlock_UUID, b'\x02' + b'\x00'*16, response=True)
> 
> I get a 0x8104

I think when you write to the unlock UUID the first byte indicates what to do. At least for my device it works like this:
- 0x02 goes to programming mode, the you can write the new key starting with 0x00
- 0x01 means that you want to unlock the device with the following code

If some bit in the second byte is set, it seems to indicate some error.
What happens when you send something to the unluckUUID also seems to depend on previous commands / errors, so make sure to powercycle your device in between tests.

userx14, 2022-07-11:
> On android every device works.
That was the reason, why ubpm only works with the faked mac...

Does that mean you have tested with a Android Phone, that has bluetooth mac address that is different from the one that was originally paired with the omron device?

In starting from Wednesday this week I should have access to a hem-7361t / m500 and can do some testing regarding the pairing process.

LazyT, 2022-07-11:
Yes, paired on tablet and works on handy too.

Tried also with cloned mac last byte +1: doesn't work on pc.

So it's not vendor specific blacklist or something like this.

userx14, 2022-07-11:
Ok, I have finished the record decoding function for hem-7361t.
[image]

Regarding the pairing, I still find it strange that sending 0x01 (16* 0x00) works for you for unlocking the device, even with the correct mac address.
When I do the same on my device I get 0x8101 (15* 0x00) indicating an error. It only works with the correct unlock code, so there seems to be some hardware level difference between the devices.

bonnybabukachappilly, 2022-07-12:
@userx14 Tried to run your code but not pairing. Return code is '820f'

userx14, 2022-07-13:
It has something to do the encryption or the pairing procedure for ble devices. Encrypted services seem to be not supported at all by bleak, therefore it is most likeley unusable for this device. The response will therefore be always zero bytes when reading a characteristic and writes will not work at all.

Since the device works with various Android phones and nRF-Connect I also tested with nRF-Connect and their nRF52840 chip using their Windows Desktop Application. There the device also returns 0x820f after connection. But when pairing the device without bonding the device suddenly works and I get the expected responses. So connecting then pairing is needed, which encrypts the btle connection and exchanges keys.

After some experimenting I was able to reproduce the connection steps with bluetoothctl on linux by manually typing the commands and it works too. So here it my log of what I typed in the shell. Not every command might be neccesary, e.g. I suspect that the btmgmt commands are optional, but please try what is required on your devices to get it to work and report your findings.

@LazyT would it be possible to "port" these commands to UBPM, since it uses the bluez stack directely?

I'm currently looking for alternative python libraries which support encrypted bluetooth le services to get this code working with the hem-7362t, but that could mean that the support for windows 10 will no longer be present.

```
#Test system
#Debian 11 - Kernel 5.10.0-15-amd64
#bluez version: 5.55-3.1
#bluez-firmware version: 1.2-4
#chipset: marvell 88W8897

#mac of omron 28:FF:B2:EE:EE:EE

#remove any previous bonds/pairings to device from the bluetooth settings
sudo su -
btmgmt connectable on
btmgmt bondable on
btmgmt io-cap 3
bluetoothctl --agent NoInputNoOutput
#start the device in p mode
scan on
#wait 2 seconds until you see at least one message from the device
scan off
connect 28:FF:B2:EE:EE:EE
pair 28:FF:B2:EE:EE:EE
menu gat
select-attribute b305b680-aee7-11e1-a730-0002a5d5c51b
notify on
write 0x02
write 0x001234123412341234123412341234123412341234
```

bonnybabukachappilly, 2022-07-13:
Which python library, I have tried Pyatt and it's good. But I am not getting exceptions and no response. while I get it on bleak. Hope it will work for you. Please keep updated.

LazyT, 2022-07-13:
> would it be possible to "port" these commands to UBPM, since it uses the bluez stack directely?

For over a year I made a bluetooth test tool and all the code was already there, with pairing.

I don't know why it wasn't working, but it works now. I don't understand it.

Didn't test reading records yet, but I guess this will work too.

Holy shit, lol...

userx14, 2022-07-13:
> For over a year I made a bluetooth test tool and all the code was already there, with pairing.
>
> I don't know why it wasn't working, but it works now. I don't understand it.
>
> Didn't test reading [...]

I would guess that they changed something with later bluez versions, btle support seems to be a recent development.
Or maybe you executed some of the btmgmt commands and they changed some state of the bt adapter?

-----
> Which python library, I have tried Pyatt and it's good.

I have looked into pygatt, but unfortunatelly it seems to be a wrapper around gatttool which is deprecated. Also gatttool when used manually did not work for me when testing.

For python it seems to work with python-bluezero with some trickery for pairing.
https://github.com/ukBaz/python-bluezero
I will push an update to the python script soon, but that means "bye bye windows support", 
unless someone finds a bluetooth librarary for BTLE which support pairing with security 1 type 2.

userx14, 2022-07-16:
Hi,
please check out the latest branch which makes it possible to pair and read data from the hem7361t.

https://github.com/userx14/omblepy/tree/hem7361test

When testing with the device I discovered it has some quirks in the protocol which differ from the hem7322t:
- only attached data sizes in multiples of (0x10) are supported for both the Eeprom read and write commands. 
- when reading memory addresses of user records which are not used / written to yet, instead of returning all 0xff the device replies with a short message without data attached to the end. I have modified the function to emulate the behavior of the hem7322 which sends all 0xff in this case.
- overwriting only the first 0x08 bytes of the currently unread records does not seem to work, probably because of the 0x10 multiple limitations in the attached data size for Eeprom read/writes. The fix would be to use two of the tx channels which the device does possess. Because of this the "-n"-mode which only reads new records is unable to reset the "unread records counter".

Additional observations:
- after the records of device has been read out once, it must be unpaired in the bluetooth settings of the os, otherwise the next pairing and connection will fail. Maybe pairing an already paired device does not work?
- bluezero or more likeley bluez in its later versions do not seem to work with the hem7322t, because some bluetooth attributes are not not available. This is the original bug I encountered with all more recent distributions than ubuntu 18.04.

bonnybabukachappilly, 2022-07-18:
I tried the new code. When I try that I am getting 
`dbus.exceptions.DBusException: org.bluez.Error.AuthenticationFailed: Authentication Failed`

userx14, 2022-07-19:
> I tried the new code. When I try that I am getting `dbus.exceptions.DBusException: org.bluez.Error.AuthenticationFailed: Authentication Failed`

With which linux distro are you running your tests?
I am testing with this Debian-11.4 KDE live iso:
https://cdimage.debian.org/debian-cd/current-live/amd64/iso-hybrid/debian-live-11.4.0-amd64-kde.iso
```
sudo apt update
sudo apt install python3-pip
pip3 install terminaltables bluezero
git clone https://github.com/userx14/omblepy/
cd omblepy
git checkout bluezeroVersion
python3.9 ./omblepy.py -p -d hem-7361t-d
```

Best,
Benjamin

userx14, 2022-07-19:
I think the the authentification bug is now fixed in the windows and crossplatform (bleak) version. 
It was a combination of pairing and reading a characteristic instead of using callbacks and assuming that delay would work there.

Toei79, 2023-06-24:
it can work with only bluetooh devices without usb ports? BP7450 , looks like it need be a sophisticated model.

Toei79, 2023-06-24:
i get this 
https://www.ebay.com/itm/113834640107

looks like one you use but , i tried on win 10 with bluettoh changer without success

userx14, 2023-06-25:
> it can work with only bluetooh devices without usb ports?
> (from your other post:) I have impression it only works on models with both usb port and bluetooth

This tool does only use bluetooth for communication, there is no usb support.
So, there is no requrirement for a devices to have an USB port to work with omblepy.

> i have those models HEM-7342T-Z and HEM 71551T-Z

HEM-7342T-Z from the outside looks simmilar to the supported HEM-7361t.
And your hem-7151t looks a bit like the supported hem-7155t.
Maybe it will work with HEM-7361t or HEM-7155t respectivly as the target device. Also I suggets using the "--loggerDebug" flag and see how far in the communication process you get. Since I do not have these devices for testing you would have to do some tests to find out on which adresses the records are stored and create a device specific file like [these ones](https://github.com/userx14/omblepy/tree/main/deviceSpecific). But I think we should open a seperate github issue for discussing device support to keep things organized.

> looks like one you use but , i tried on win 10 with bluettoh changer without success

Changing the mac address of the bluetooth adapter is only required for [ubpm](https://codeberg.org/lazyt/ubpm), while omblepy shoud include the neccesary pairing routine to work without altering the mac address. 
The stick you bought uses the special csr harmony driver on windows and from my testing I was NOT able to get it to work with any omron device under Windows.
The stick is recommended by UBMP because it is one of the few bluetooth sticks where you can modify the mac-address under linux, but omblepy does not need this feature and this stick does not work on windows. On Linux, you have to use a specific version of the bluetooth stack and other requirements, so it's quiet cumbersome.
Therefore if you are using windows I recommend using a different bluetooth chipset which does not require sepecial bluetooth stacks, maybe you already have a builtin bluetooth module in some laptop available.

Toei79, 2023-06-25:
> This tool does only use bluetooth for communication, there is no usb support. So, there is no requrirement for a devices to have an USB port to work with omblepy.
>
>
> HEM-7342T-Z from the outside [...]

hey thanks for deep answer , i tried script but no dice,   but i gonna check with debug part on linux to see what shows.

i have two usb bluetoth , so this one doesnt works , but it was that path what usb bluetoth  you recommend? i cant know about chipset at least online, for what i understand,  im in usa. 

thats means there its more chances with old laptop to get it work? i mean with the clone mac addreses, kinda like  trial and error thing?

userx14, 2023-06-25:
> i tried script but no dice, but i gonna check with debug part on linux to see what shows.

This debugging feature is also available on Windows.
I would expect the script to fail at some stage, because I suspect that the blood pressure records are stored at different addresses.

> what usb bluetoth you recommend? 

If you are on windows please try to use an bluetooth dongle / builtin card without some special bluetooth stack. Intel / Marvell / Realtek chipsets worked for me. 

I only came across one single dongle that does not work on Windows, it was based on a csr8510 and required installing the CSR Harmony Stack. I think the ebay link you sent is exactely this csr stick which will not work with omblepy on Windows.

> i mean with the clone mac addreses, kinda like trial and error thing?

The mac address cloning / changing step is NOT needed for omblepy.

Toei79, 2023-06-26:
ok i tried this bluetooth device and other one one in the pic.  i did paring mode on omron device. 

[image]

[image]

gotcha what you say them they dont documented too much about just mention the three bluetooth devices ,  i never know about linux only. 

i have a laptop from 2012 im going to give a try there.

userx14, 2023-06-26:
Hi, the error in your first screenshot is not related to bluetooth communication. 

It has occured because the tool uses the device name after the -d flag to load the device specific programm file with this name from [here](https://github.com/userx14/omblepy/tree/main/deviceSpecific). So if run it with "-d hem-7342" it searches for hem-7342.py there and aborts.
I have so far only determined the addresses for four omron devices, but some of them are quiet simmilar.

Therefore my recommendation from the previous post was to just pass as one of the supported devices which is most simmilar to your device after the -d flag: (-d HEM-7361t).

Regarding the second screenshot, you do not need to change the mac address for omblepy.

Toei79, 2023-06-26:
I get this

[image]

Toei79, 2023-06-26:
ok i press bluetooh button then i get this

[image]

Toei79, 2023-06-26:
i move to windows and i connected

[image]

nice now i need try with ubpm , thats easy? im reading here and see

Toei79, 2023-06-26:
i able to connect but after i get this
[image]
without the -p , at first time i get csv after this

userx14, 2023-06-26:
Hi,

nice, looks like the initial pairing worked.

But when you did the second test without the -p flag it has lost connection to the device. 

> try again and dont show up here then a display dialog appear from bluetooth on windows i click accept 

Have you enabled the normal bluetooth communication by shortly pressing the bluetooth button on your omron for this test? A small o should be displayed on the omron's screen.
Also the timeout of the omron is not that long, meaning you have to relatively quickly connect after enabling the bluetooth mode.
You can also try some of the troubleshooting steps from the readme, like unpairing the device in the windows bluetooth settings, the bluetooth pairing is independent from the omron pairing procedure (-p), so you do not have to use the -p option from now on. 

> nice now i need try with ubpm , thats easy?

Well I think the device is also not supported by ubpm.
Ubpm and omblepy are simmilar tools, with ubpm having a nice gui and database to store the records, while omblepy aims to be a simple command line tool to just dump the records straight to a csv file.

> ubpm i started to import data, csv dont compatible with ubpm.

Most likeley, no records have been sent yet, so the csv file is expected to be empty. Probably that's why ubpm rejects reading it.

To see some more info could you add the `--loggerDebug` option when running the tool?

Toei79, 2023-06-26:
ok when the first tries that no happened and runs normal even without -p , when i did i able to get .csv files one with some data that i can see on excel but not on ubpm .  

maybe get those erratics because i was searching on bluetooth windows? maybe? that was after , just see bp250 i guess on windows bluetooth so i tought was fine, 

i keep trying and run script then get those erratics on command line, but at same time again a popup appears the windows bluetooth showing connect to bp250 , what i did, at same time i have ubpm open then it find the blood pressure and donwload all data, and display all stuff perfectly

Toei79, 2023-06-27:
im going check with debug tonight, its that normal or its not fine?
yea i press bluetooth button on omron device

if keep getting trouble i do those steps on readme

[user1.csv](https://github.com/userx14/omblepy/files/11878070/user1.csv)

this its file that i get from omron , but dont load fine on ubpm.

userx14, 2023-06-27:
> im going check with debug tonight, its that normal or its not fine? yea i press bluetooth button on omron device
>
> if keep getting trouble i do those steps on readme [...]

Oh, I just remembered that ubpm does not need the .csv file, it needs ubpm.json which should also be generated by omblepy automatically. The csv is for using something like Excel or the plotCsv.py visializer from omblepy.

> when its pairing by windows its not keep for longer

I'm not really sure what you meant by that.

The bluetooth connection process is a bit finicky, for me it works best if the device is not paired / stored in the windows bluetooth settings to start. You can use the "-m" option with your omron's bt-mac address to skip the scanning for the device, which makes it easier to stay within the timeout of the onron for connecting. Only accept the bluetooth pairing popup dialog from windows when it shows up directely after starting the tool.

Toei79, 2023-06-28:
[image]
[image]

with debug .  i tried with HEM 7342

Toei79, 2023-06-28:
Model HEM7150T
looks working 
[image]

Toei79, 2023-06-28:
i tried connect on linux but i get the same error, i dont connect nothing yet on linux, so i tried there
[image]

Toei79, 2023-06-28:
sorry for the mess here lmk if i deleted all this messages, i just reading all messages, and then try open one.

Toei79, 2023-06-29:
i have other omron device, one with ekg reading , model its BP7900 , HEM-7530T-Z

i rereading here, its a way to get needed info to runs those with omblepy? so far i have two devices

userx14, 2023-06-29:
Ok, have I understood it right that:
- your hem7151T works with the code of the 7155T
- your hem7342T refuses to pair

maybe you can try to alter the protection-level for the pairing in [this line](https://github.com/userx14/omblepy/blob/3b49c009f0579487660e534aba98d4602b291d60/omblepy.py#L364). Maybe protection level 1 will work, it is possible that this device has some different pairing requirements, at least thats what I found on bleaks github when searching for error 19 when pairing.

- you want to get data (including ekg) from the hem-7530t

As a first attempt you can try if pairing works. Because this device is not very simmilar to the currently supported devices, it will likeley require some reverse engineering. 

You can record all in- and outgoing bluetooth communication of Android phones, the relevant search term would be "btsnoop_hci.log". Some newer android devices make it more difficult to access the logged data though. The recorded captured data can then be inspected with Wireshark to analyze the sent packets.

It is possible that I can find out some address offsets for this device aswell, but currently I do not have that much time available, probably I will check on the weekend.

Toei79, 2023-06-29:
same i gonna check weekend , but yea just read and reread to familiarize  with this.  i have old android phones.  idk if that helps and one note .

Toei79, 2023-06-29:
ok i tried this i read here you say

''I always expected that there is a normal bluetooth connection mode, like there is with the hem-7322t, where you short press the blutooth button, and it goes into a mode where the little square directely appears and not the P''

both devices have that instead a p , with short press i get like a small o? then i tried run  script, with 2 and 1 on both none works.

userx14, 2023-06-29:
> change to 1 and not worked.

Yeah changing this to 1 was what I hoped could fix the problem. 

> ok i tried this i read here you say
> 
> ''I always expected that there is a normal bluetooth connection mode, like there is with the hem-7322t, where you short press the blutooth button, and it goes into a mode where the little square directely appears and not the P''
> 
> both devices have that instead a p , with short press i get like a small o? then i tried run script, with 2 and 1 on both none works.
> 
> Ubuntu 22.04.2 LTS

Your omron devices should be in pairing mode "blinking P / longpress" when you run omblepy with the -p option. 
For normal data readout without the -p flag, it should be in the "little square / shortpress" mode. It is possible that this is a newer model and the pairing mode is different. Have you tested pairing with windows?

For your hem-7530t I have put my best guess for the addresses in [this branch](https://github.com/userx14/omblepy/blob/hem-7530t-testing/deviceSpecific/hem-7530t.py). But for the internal structure of the records I'm not so sure. If you can get the bytes of the records printed and have the data to compare to we might be able to deduce where which value is stored.

Toei79, 2023-06-30:
all models i have,  comes with both type of pairing mode short and long
i tried short press then without the -p still get error. 

i tried but not worked.
[image]

userx14, 2023-06-30:
What bluetooth adapter are you currently using?
I think some of the newer pairing features are only supported with later bluetooth versions (if I read this correctely BT LE Secure connections were added with [Bluetooth 4.2](https://www.bluetooth.com/blog/bluetooth-pairing-part-1-pairing-feature-exchange/) ) and it could be possible that omron relies on that.

Things you could test:
Does pairing work with the windows os dialog?
Does the pairing work on Android (it has a more reliable bluetooth stack in my experience)?
If pairing on android works there is an app called nRF Connect, that can be used to manually read / write bluetooth characteristics and manually do some of the transmissions.

Toei79, 2023-06-30:
its very old i guess i get it like 4 years ago idk if its bluetooth 4.2

[image]
its this one kinovo brand. 

its ok get a 5.0? one with latest stuff?

userx14, 2023-07-01:
> its very old i guess i get it like 7 years ago idk if its bluetooth 4.2
> 
> [image] its this one kinovo brand.
> 
> https://www.amazon.com/Kinivo-USB-Bluetooth-4-0-Compatible/dp/B007Q45EF4/ref=sr_1_1_sspa?crid=5VZFI1XF065A&keywords=kinivo+bluetooth+adapter&qid=1688157068&sprefix=kinovo+blue%2Caps%2C105&sr=8-1-spons&sp_csd=d2lkZ2V0TmFtZT1zcF9hdGY&psc=1
> 
> its ok get a 5.0? one with latest stuff?

I would first try if pairing with android works before buying something new.
I think most phones use quiet new bluetooth versions. 
If the pairing works using the android settings bluetooth menu and you can see bluetooth characteristics with nRF connect, it is probably related to the bluetooth version, or it could be a bug with the bluetooth library (bleak) I'm using.

I think your stick uses a Broadcom BCM20702, which only supports up to bluetooth 4.0 .

Did you test pairing just using the windows os dialog without omblepy using this bt stick?

Toei79, 2023-07-01:
oh what that means connect bt stick to android phone?

i have this stick and the other one that i bought , only one working its this one,  i able to succesfully paring on windows and linux with this bt stick, one of my three omron devices.

userx14, 2023-07-02:
> oh what that means connect bt stick to android phone?
> 
> i have this stick and the other one that i bought , only one working its this one, i able to succesfully paring on windows and linux with this bt stick, one of my three omron devices.

No, just use the bluetooth built into your phone. I was hoping the hardware in the phone supports the newer bluetooth standard, just to test if it makes a difference for the devices that do not workin with your stick.

Toei79, 2023-07-03:
> No, just use the bluetooth built into your phone. I was hoping the hardware in the phone supports the newer bluetooth standard, just to test if it makes a difference for the devices that do not [...]

i dont get it , i mean, phone connects successfully with omron device with all three 

or its connect phone with bluetooh usb device?

userx14, 2023-07-03:
The idea was to use the Android Os Pairing dialog on a second phone which does not have the omron app installed and was not the last phone paired to the device. This would create a simmilar scenario to a pc trying to pair, just that most phones have hardware supporting more recent bluetooth versions.

Anyway I think the most likeley cause is just the too low bluetooth version of your usb adapter.

Toei79, 2023-07-03:
> The idea was to use the Android Os Pairing dialog on a second phone which does not have the omron app installed and was not the last phone paired to the device. This would create a simmilar scenario to a pc trying to pair, just that most phones have hardware supporting more recent bluetooth versions. If pairing with the phone works it is most likeley just the bluetooth version of your usb adapter causing the probem. If it does not work, onron likeley accesses some bluetooth attributes before pairing, but this is kind of unlikely.
> 
> Anyway I think the most likeley cause is just the too low bluetooth version of your usb adapter. So if you can test on windows device with a chipset / usb dongle with more recent bluetooth version, just try if it works 😃 .

i tried with a tablet that never connected

[image]

and both devices connected

userx14, 2023-07-04:
> both devices connected

Ok, so I would say it is most likely that the reason the pairing fails on windows is your usb dobgle's missing support for newer bluetooth features (4.2).

> but on omron devices keep blinking P

The continued blinking of the P on the screen is normal, one needs to write a pairing key to some bluetooth characteristics for it to stop, which is what omblepy would do after the pairing succeeded.

Toei79, 2023-07-04:
> Ok, so I would say it is most likely that the reason the pairing fails on windows is your usb dobgle's missing support for newer bluetooth features (4.2).
>
>
> The continued blinking of the P on the [...]

ok what usb dongle you recommend? latest its one 5.0 or 5.1? that one works

Toei79, 2023-07-04:
i guess i try get one on ebay , but i only see 5.0 so far.

Toei79, 2023-07-04:
i get one from amazon 5.3 , it comes thursday , i update after i try

Toei79, 2023-07-04:
im trying use tablet , but idk how do it 

i install it pydroid 3 , termius , i get omblepy file on the tablet but idk how run it.

[19 more replies]

closed 2023-07-25 by userx14