userx14/omblepy #4: Recovering device's internal clock

vulcainman, 2022-09-28:
Thanks for your nice project, it gave me a lot of information about Omron devices' Bluetooth protocol. I would like to implement other features in relation with devices' internal clock (how to recover it, to reconfigure it, etc...) and I've several question :

1. Do you have any information concerning devices' internal clock ?
2. How did you get EEPROM address and data format for measurement information (official documentation reverse engineering, communication analysis, EEPROM diff analysis) ?

Thanks !

userx14, 2022-09-28:
To 1)
Yes I do, but if I remember correctely this is device specific. 
For example with HEM7361 the address for clock writes is most likeley at 0x44 and for writes at 0x88
```
read 0x0044 16010d0c000328000091
```
This example reads the current device time, which is (0x16) = year 2022, (0x01) = month 01, (0x0d) = day 13 and so on.
HEM7361 also performs this read/write of the current time for every observed communication with my android phone.

To 2)

official documentation -> no, have you found any? 😃 

reverse engineering -> yes analyzing certain apk files can be extremely helpful for reverse engineering, since they often contain information about all supported devices. I have a semi-automated process for this (see later part).

communication analysis -> works well, to log bt communication with android retrieve <a href="https://medium.com/@charlie.d.anderson/how-to-get-the-bluetooth-host-controller-interface-logs-from-a-modern-android-phone-d23bde00b9fa"> hcisnoop.log</a> and use <a href="https://www.wireshark.org/download.html"> wireshark</a> with filter "btatt". Might only work on some devices and obtaining the log is easyer with a rooted phone.

EEPROM diff analysis -> no, could be done but sounds like too much work, since the previous two option should give you enough information. I have to admit that I have tried that tough, with the intention to just add a microcontroller on the board and skip the bluetooth communication alltogether. But thats unfortunatelly easier said than done:
You could log the writes on the bus lines while the device is working. But reading the eeprom between these writes (or when the device is powered off) would require you to inject power which also powers the main processor which pulls the eeprom bus lines to some logic levels, interfering with your read commands. So cutting the power to the processor or switching the bus lines would have been required, and I didn't want to risk the device, since I got bluetooth working reliabelly around the same time.

---

Do you want get this working for one specific device or all devices?

If you want to build a solution working with all devices you might find some use for a little python script which can extract some device specific information from a certain apk file. The goal was to autogenerate these device specific configuration files with all memory locations, read formats, and so on, but it's not finished.
Currently extracting per device adresses and offsets is somewhat working.
I apologize in advance for the little stalking, but is this your <a href="https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=817277" >mail</a> or could you provide me one, if you have interest in the tool?

Best,
Benjamin

userx14, 2022-09-28:
Hi @vulcainman,

on hem-7600 it should be a read around 0x276, and the first byte is the year+2000, the second byte the month and so on.
Deleted the comment to protect personal info, check your mail 😄, thanks for the info.

Best,
Benjamin

userx14, 2022-10-18:
For future reference the clock sync is now included in the HEM-7322 driver in the current development branch
https://github.com/userx14/omblepy/blob/dc6441313a76f58fd9597c8cc902c1702078851e/deviceSpecific/hem-7322t.py#L39-L54

closed 2022-10-18 by userx14