userx14/omblepy #5: 7600T and Missing Functionality

RoninTech, 2022-10-18:
Hi,

I have used your (main branch) omblepy.py to successfully pair to my 7600T using "-d hem-7322T-d -p".  It paired on the second attempt.  I would like to help with getting the data read working and any other possible features.  I am doing this on a raspberry pi 4 with the latest 64bit debian bullseye.

What needs doing to get this working? 

Thanks

RoninTech, 2022-10-18:
Here is the output I receive when trying to read data after copying the 7600T device file from the bluez branch.  It appears to stop as expected when referencing the missing getNewRecords function:
```
python ./omblepy.py -d HEM-7600T
2022-10-18 17:47:14,859 - omblepy-logger - INFO - Attempt to import module for device hem-7600t
To improve your chance of a successful connection please do the following:
 -remove previous device pairings in your OS's bluetooth dialog
 -enable bluetooth on you omron device and use the specified mode (pairing or normal)
 -do not accept any pairing dialog until you selected your device in the following list

Select your Omron device from the list below...
+----+-------------------+-------------------------------+------+
| ID | MAC               | NAME                          | RSSI |
+----+-------------------+-------------------------------+------+
| 0  | 28:FF:B2:XX:XX:XX | BLESmart_0000034028XXXXXXXXXX | -58  |
+----+-------------------+-------------------------------+------+
Enter ID or just press Enter to rescan.
0
Attempt connecting to 28:FF:B2:ED:65:C9.
/home/xxx/omblepy/./omblepy.py:332: FutureWarning: This method will be removed future version, use the services property instead.
  services = await bleClient.get_services()
2022-10-18 17:47:35,147 - omblepy-logger - INFO - requesting records
2022-10-18 17:47:35,147 - omblepy-logger - INFO - Unpair and disconnect
Traceback (most recent call last):
  File "/home/xxx/omblepy/./omblepy.py", line 349, in <module>
    asyncio.run(main())
  File "/usr/lib/python3.9/asyncio/runners.py", line 44, in run
    return loop.run_until_complete(main)
  File "/usr/lib/python3.9/asyncio/base_events.py", line 642, in run_until_complete
    return future.result()
  File "/home/xxx/omblepy/./omblepy.py", line 342, in main
    allRecs = await deviceSpecific.getNewRecords(bluetoothTxRxObj, UseAndResetUnreadCounter = args.newRecOnly)
AttributeError: module 'hem-7600t' has no attribute 'getNewRecords'
```
After pairing I also tried these:
```
gatttool -b 28:FF:B2:ED:65:C9 --primary
attr handle = 0x0001, end grp handle = 0x0051 uuid: 00001800-0000-1000-8000-00805f9b34fb
attr handle = 0x0100, end grp handle = 0x0112 uuid: 00001801-0000-1000-8000-00805f9b34fb
attr handle = 0x0200, end grp handle = 0x0281 uuid: 0000180a-0000-1000-8000-00805f9b34fb
attr handle = 0x0300, end grp handle = 0x03a2 uuid: ecbe3980-c9a2-11e1-b1bd-0002a5d5c51b
attr handle = 0x0400, end grp handle = 0x0421 uuid: 00001810-0000-1000-8000-00805f9b34fb
attr handle = 0x0500, end grp handle = 0x0512 uuid: 0000180f-0000-1000-8000-00805f9b34fb
attr handle = 0x0600, end grp handle = 0x0612 uuid: 00001805-0000-1000-8000-00805f9b34fb
```

```
gatttool -b 28:FF:B2:ED:65:C9 --characteristics
handle = 0x0010, char properties = 0x02, char value handle = 0x0011, uuid = 00002a00-0000-1000-8000-00805f9b34fb
handle = 0x0020, char properties = 0x02, char value handle = 0x0021, uuid = 00002a01-0000-1000-8000-00805f9b34fb
handle = 0x0030, char properties = 0x0e, char value handle = 0x0031, uuid = 00002a02-0000-1000-8000-00805f9b34fb
handle = 0x0040, char properties = 0x02, char value handle = 0x0041, uuid = 00002a03-0000-1000-8000-00805f9b34fb
handle = 0x0050, char properties = 0x02, char value handle = 0x0051, uuid = 00002a04-0000-1000-8000-00805f9b34fb
handle = 0x0110, char properties = 0x20, char value handle = 0x0111, uuid = 00002a05-0000-1000-8000-00805f9b34fb
handle = 0x0210, char properties = 0x02, char value handle = 0x0211, uuid = 00002a23-0000-1000-8000-00805f9b34fb
handle = 0x0220, char properties = 0x02, char value handle = 0x0221, uuid = 00002a24-0000-1000-8000-00805f9b34fb
handle = 0x0230, char properties = 0x02, char value handle = 0x0231, uuid = 00002a25-0000-1000-8000-00805f9b34fb
handle = 0x0240, char properties = 0x02, char value handle = 0x0241, uuid = 00002a26-0000-1000-8000-00805f9b34fb
handle = 0x0250, char properties = 0x02, char value handle = 0x0251, uuid = 00002a27-0000-1000-8000-00805f9b34fb
handle = 0x0260, char properties = 0x02, char value handle = 0x0261, uuid = 00002a28-0000-1000-8000-00805f9b34fb
handle = 0x0270, char properties = 0x02, char value handle = 0x0271, uuid = 00002a29-0000-1000-8000-00805f9b34fb
handle = 0x0280, char properties = 0x02, char value handle = 0x0281, uuid = 00002a2a-0000-1000-8000-00805f9b34fb
handle = 0x0310, char properties = 0x1e, char value handle = 0x0311, uuid = b305b680-aee7-11e1-a730-0002a5d5c51b
handle = 0x0320, char properties = 0x0c, char value handle = 0x0321, uuid = db5b55e0-aee7-11e1-965e-0002a5d5c51b
handle = 0x0330, char properties = 0x0c, char value handle = 0x0331, uuid = e0b8a060-aee7-11e1-92f4-0002a5d5c51b
handle = 0x0340, char properties = 0x0c, char value handle = 0x0341, uuid = 0ae12b00-aee8-11e1-a192-0002a5d5c51b
handle = 0x0350, char properties = 0x0c, char value handle = 0x0351, uuid = 10e1ba60-aee8-11e1-89e5-0002a5d5c51b
handle = 0x0360, char properties = 0x12, char value handle = 0x0361, uuid = 49123040-aee8-11e1-a74d-0002a5d5c51b
handle = 0x0370, char properties = 0x12, char value handle = 0x0371, uuid = 4d0bf320-aee8-11e1-a0d9-0002a5d5c51b
handle = 0x0380, char properties = 0x12, char value handle = 0x0381, uuid = 5128ce60-aee8-11e1-b84b-0002a5d5c51b
handle = 0x0390, char properties = 0x12, char value handle = 0x0391, uuid = 560f1420-aee8-11e1-8184-0002a5d5c51b
handle = 0x03a0, char properties = 0x10, char value handle = 0x03a1, uuid = 8858eb40-aee8-11e1-bb67-0002a5d5c51b
handle = 0x0410, char properties = 0x20, char value handle = 0x0411, uuid = 00002a35-0000-1000-8000-00805f9b34fb
handle = 0x0420, char properties = 0x02, char value handle = 0x0421, uuid = 00002a49-0000-1000-8000-00805f9b34fb
handle = 0x0510, char properties = 0x12, char value handle = 0x0511, uuid = 00002a19-0000-1000-8000-00805f9b34fb
handle = 0x0610, char properties = 0x1a, char value handle = 0x0611, uuid = 00002a2b-0000-1000-8000-00805f9b34fb
```

userx14, 2022-10-18:
Hi @RoninTech,

currently I'm in the process of restructuring the code to make implementing device specific drivers simpler.
It is still in the beta testing phase, but here is the branch:
https://github.com/userx14/omblepy/tree/sharedDriverTest

Because I don't have this specific device I got some help form @vulcainman, he has just today uploaded, as far as I know, a fully working driver for this device:
https://github.com/vulcainman/omblepy/tree/sharedDriverTest

This branch with bleak should also work with linux, but I haven't tested that yet.
If you need a bluezero version please let me know.

Best,
Benjamin

vulcainman, 2022-10-18:
Hi @RoninTech ,

As @userx14 told you, the device specific for HEM-7600T I've uploaded this morning in should be fully functionnal (tested on GNU/Linux Debian Bookworm). Give it a try and tell us if it works as expected on your system.

RoninTech, 2022-10-18:
Thanks @userx14 and @vulcainman!  Worked first time. 😄 

Now to setup the DB and Grafana.

userx14, 2022-10-18:
> Thanks @userx14 and @vulcainman! Worked first time. 😄
> 
> Now to setup the DB and Grafana.

Always feel free to fork the repo, a better visualization than the rudamentary "[plotCsv.py](https://github.com/userx14/omblepy/blob/main/plotCsv.py)" sounds interesting.
If you haven't seen it yet, lazyT has an <a href="https://codeberg.org/LazyT/ubpm">interesting project (ubpm)</a> with more plotting features and a database.

RoninTech, 2022-10-19:
> Always feel free to fork the repo, a better visualization than the rudamentary "[plotCsv.py](https://github.com/userx14/omblepy/blob/main/plotCsv.py)" sounds interesting. If you haven't seen it yet, lazyT has an [interesting project (ubpm)](https://codeberg.org/LazyT/ubpm) with more plotting features and a database.

Forked.  I did see ubpm but my only machine is a low powered arm device and I don't want to compile stuff on it if I don't have to.  I'm already running influxdb and grafana dockers for other projects (bluetooth scale).  Plus I enjoy playing with this stuff.

Thanks again for the great work on this from both of you.  I really lucked out with the timing. :smiley:

closed 2022-10-26 by userx14