userx14/omblepy #12: OMRON RS7: bleak.exc.BleakDBusError: [org.bluez.Error.AuthenticationFailed] Authentication Failed

baderas, 2023-04-25:
When trying to pair my OMRON RS7 i always get an error: bleak.exc.BleakDBusError: [org.bluez.Error.AuthenticationFailed] Authentication Failed.  

I tried all troubleshooting steps and maybe it helps: i can connect, trust and pair it from within bluetoothctl.

Any ideas what I could try?

```
$ python3 ./omblepy.py -p -d HEM-7361T --mac XX:XX:XX:XX:XX:XX --loggerDebug
2023-04-25 17:43:33,413 - omblepy - INFO - Attempt to import module for device hem-7361t
2023-04-25 17:43:33,433 - omblepy - INFO - Attempt connecting to XX:XX:XX:XX:XX:XX.
2023-04-25 17:43:39,968 - omblepy - INFO - unpair and disconnect
Traceback (most recent call last):
  File "/tmp/omblepy/./omblepy.py", line 364, in main
    await bleClient.pair(protection_level = 2)
  File "/usr/lib/python3.10/site-packages/bleak/__init__.py", line 555, in pair
    return await self._backend.pair(*args, **kwargs)
  File "/usr/lib/python3.10/site-packages/bleak/backends/bluezdbus/client.py", line 466, in pair
    assert_reply(reply)
  File "/usr/lib/python3.10/site-packages/bleak/backends/bluezdbus/utils.py", line 22, in assert_reply
    raise BleakDBusError(reply.error_name, reply.body)
bleak.exc.BleakDBusError: [org.bluez.Error.AuthenticationFailed] Authentication Failed

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/tmp/omblepy/./omblepy.py", line 389, in <module>
    asyncio.run(main())
  File "/usr/lib/python3.10/asyncio/runners.py", line 44, in run
    return loop.run_until_complete(main)
  File "/usr/lib/python3.10/asyncio/base_events.py", line 649, in run_until_complete
    return future.result()
  File "/tmp/omblepy/./omblepy.py", line 387, in main
    await bleClient.disconnect()
  File "/usr/lib/python3.10/site-packages/bleak/__init__.py", line 540, in disconnect
    return await self._backend.disconnect()
  File "/usr/lib/python3.10/site-packages/bleak/backends/bluezdbus/client.py", line 413, in disconnect
    assert self._bus is None
AssertionError
```

userx14, 2023-04-25:
Hi @baderas 

Do you get the a pairing dialog right when the python script tries to connect?
If you have the possibility to test with a windows device please do so. From experience the windows bluetooth stack seems to work niceley with omron devices. This way we might be able to find out if the pairing issue is caused by your specific omron device vs your os and setup.
I also had great trouble getting the omron devices to work on linux and bluez. Only some distros and versions of bluez seem to work, I had the best results with Debian 11.4.0 Kde https://cdimage.debian.org/debian-cd/current-live/amd64/iso-hybrid/debian-live-11.4.0-amd64-kde.iso

Did you already try some suggestions from troubleshooting section in the readme https://github.com/userx14/omblepy#troubleshooting ?

Best,
Benjamin

baderas, 2023-04-29:
Hi Benjamin,

I tried it with windows, the paring with omblepy worked. The data however could not be received by omblepy (what was to be expected due to no support of the RS7 datastructure).
But on windows I then had problems with ubpm, which did not work. It did not find any devices with my bt stick.
I tried the same on linux, It did connect and retrieve device, name, version, etc., but it did not receive data, the error was something like "no answer received". 
Then I bought another bt stick, one from here (macwiedis comment): https://codeberg.org/LazyT/ubpm/issues/26

The result on both OS were the same with the new stick. Nice :-/

But then I poked around and tried many things to get ubpm to receive the blood pressure data.
I was succesful on linux by connecting to my omron RS7 with the OS bluetooth app.
Now it can import data, but mostly on second try, first try often aborts with "no answer received".

After that I discovered I could even pair with omblepy using the following method on linux:
- Connect to Omron RS7 with OS bluetooth app
- put the bt usb stick in and out
- ./omblepy.py -p -d HEM-7361T --mac XX:XX:XX:XX:XX:XX --loggerDebug
- omblepy shows some error at the end, but thats just a ignorable timeout when disconnecting.
- Omron is now succesfully paired.
- put the bt usb stick in and out
- Connect to Omron RS7 with OS bluetooth app
- ubpm is working (mostly on second try as described above)

I did not manage to get ubpm on Windows to work, I tried many things, but somehow it never finds suitable devices when searching for devices.
Both usb bt sticks behaved basically the same, so my old one is as good as my new one.

I close this issue now.

closed 2023-04-29 by baderas

userx14, 2023-04-29:
Hi @baderas ,

the bluetooth sticks mentioned on the UBPM page are selected because one can alter their bluetooth-mac address. This way one can pair the Omron device with the official Android omron app, then clone the mac address of the Android phone to the usb dongle. This used to be the only method to get any data from the devices. But omblepy supports the pairing sequence and is therefore able to connect and pair with a bluetooth adapter with it's original mac and UBPM will work afterwards.

> The data however could not be received by omblepy (what was to be expected due to no support of the RS7 datastructure).

If you are interested in support for the RS7 / HEM-6232T-D in omblepy please let me know. 
I have attempted to include the information from ubpm, but cannot guarantee it will work and can't test it: https://github.com/userx14/omblepy/tree/hem-6232t-testing .

> I did not manage to get ubpm on Windows to work, I tried many things, but somehow it never finds suitable devices when searching for devices.

Can't really help with that, have you asked on their development site already?

Best,
Benjamin

userx14, 2023-05-12:
Thanks for testing.
I hope that I've fixed the year and dia pressure, a new commit is available on the [hem6232t-testing](https://github.com/userx14/omblepy/tree/hem-6232t-testing) branch.
If you have time, please give it a try.

baderas, 2023-05-18:
I tried it, dia pressure is now fine, but the year is still wrong by 2 years:
wrong: 2021-05-18 13:00:52
correct: 2023-05-18 13:00:52

userx14, 2023-05-20:
Is it wrong for all of your records?

Because I though I would have fixed it by discarding the upper two bits:
`((2087-2000) & 0x3f)+2000 = 2023`
`((2146-2000) & 0x3f)+2000 = 2018`
`((2151-2000) & 0x3f)+2000 = 2023`
and I don't have a good explanation why the change did not fix it.