userx14/omblepy #13: Which usb bt adapter?

symgryph, 2023-05-09:
There are a myriad of them, and the ons I haveCSR 4.0 don't work. Whats a good one to use? The system pairs, but then never finishes.

userx14, 2023-05-09:
Hi @symgryph ,

for windows almost all bluetooth sticks / cards I came across did work with omblepy. But there is one exception:
I had a dongle which came with an adittional bluetooth stack called "csr harmony" which I could not get to work on windows. I'm not sure, but it is possible that your csr stick has the same chipset. The same CSR8510 based stick does however work on some versions of linux.

In general I would recommend just to try different bluetooth dongles / integrated wifi bluetooth combo cards that you already own. 

Can you give some more details on what exactly "does never finish"? Do you have some log output from running the tool with the --loggerDebug flag described in the readme? Are you using an already supported omron device?

Best,
Benjamin

Toei79, 2023-06-24:
it can work with bp50 model? i have impression it only works on models with both usb port and bluetooth

in my case i only can use bluetooth on this one

closed 2023-06-29 by userx14