userx14/omblepy #22: HEM-7600T: works only when bluetoothctl is running

shalak, 2024-02-14:
When I tried pairing and connecting to HEM-7600T, I kept getting the following error:

```
# python omblepy.py -d HEM-7600T --mac XX:XX:XX:XX:XX:XX --loggerDebug
2024-02-14 16:51:26,089 - omblepy - INFO - Attempt to import module for device hem-7600t
2024-02-14 16:51:26,134 - omblepy - INFO - Attempt connecting to XX:XX:XX:XX:XX:XX.
2024-02-14 16:51:32,465 - omblepy - INFO - unpair and disconnect
Traceback (most recent call last):
  File "/srv/services/omble/omblepy/omblepy.py", line 364, in main
    await bleClient.pair(protection_level = 2)
  File "/srv/services/omble/venv-omble/lib/python3.11/site-packages/bleak/__init__.py", line 629, in pair
    return await self._backend.pair(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/srv/services/omble/venv-omble/lib/python3.11/site-packages/bleak/backends/bluezdbus/client.py", line 473, in pair
    assert_reply(reply)
  File "/srv/services/omble/venv-omble/lib/python3.11/site-packages/bleak/backends/bluezdbus/utils.py", line 22, in assert_reply
    raise BleakDBusError(reply.error_name, reply.body)
bleak.exc.BleakDBusError: [org.bluez.Error.AuthenticationFailed] Authentication Failed

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/srv/services/omble/omblepy/omblepy.py", line 389, in <module>
    asyncio.run(main())
  File "/usr/lib/python3.11/asyncio/runners.py", line 190, in run
    return runner.run(main)
           ^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/asyncio/runners.py", line 118, in run
    return self._loop.run_until_complete(task)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/asyncio/base_events.py", line 653, in run_until_complete
    return future.result()
           ^^^^^^^^^^^^^^^
  File "/srv/services/omble/omblepy/omblepy.py", line 387, in main
    await bleClient.disconnect()
  File "/srv/services/omble/venv-omble/lib/python3.11/site-packages/bleak/__init__.py", line 614, in disconnect
    return await self._backend.disconnect()
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/srv/services/omble/venv-omble/lib/python3.11/site-packages/bleak/backends/bluezdbus/client.py", line 420, in disconnect
    assert self._bus is None
           ^^^^^^^^^^^^^^^^^
AssertionError
```

In the meantime, `journalctl -u bluetooth.service` showed:

```
lut 14 16:51:17 hostname bluetoothd[3979]: src/device.c:new_auth() No agent available for request type 2
lut 14 16:51:17 hostname bluetoothd[3979]: device_confirm_passkey: Operation not permitted
```

However, whenever I start `bluetoothctl` in other console windows (and keep it running), it does suceed to pair/sync, the `user1.csv` contains correct data and `journalctl` stays silent.

But still, it ends with an `AssertionError`:

```
# python omblepy.py -d HEM-7600T --mac XX:XX:XX:XX:XX:XX
2024-02-14 16:56:14,429 - omblepy - INFO - Attempt to import module for device hem-7600t
2024-02-14 16:56:14,468 - omblepy - INFO - Attempt connecting to XX:XX:XX:XX:XX:XX.
2024-02-14 16:56:21,527 - omblepy - INFO - communication started
2024-02-14 16:56:22,436 - omblepy - INFO - start reading data, this can take a while, use debug flag to see progress
2024-02-14 16:56:29,049 - omblepy - INFO - communication finished
2024-02-14 16:56:29,056 - omblepy - INFO - writing data to user1.csv
2024-02-14 16:56:29,062 - omblepy - INFO - unpair and disconnect
Traceback (most recent call last):
  File "/srv/services/omble/omblepy/omblepy.py", line 389, in <module>
    asyncio.run(main())
  File "/usr/lib/python3.11/asyncio/runners.py", line 190, in run
    return runner.run(main)
           ^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/asyncio/runners.py", line 118, in run
    return self._loop.run_until_complete(task)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/asyncio/base_events.py", line 653, in run_until_complete
    return future.result()
           ^^^^^^^^^^^^^^^
  File "/srv/services/omble/omblepy/omblepy.py", line 387, in main
    await bleClient.disconnect()
  File "/srv/services/omble/venv-omble/lib/python3.11/site-packages/bleak/__init__.py", line 614, in disconnect
    return await self._backend.disconnect()
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/srv/services/omble/venv-omble/lib/python3.11/site-packages/bleak/backends/bluezdbus/client.py", line 420, in disconnect
    assert self._bus is None
           ^^^^^^^^^^^^^^^^^
AssertionError
```

I'm on Ubuntu 23.04 & bluetoothctl: 5.66, commit 1f43fefc835f5b40033b0df1d39525060bb8c096.

Is there anything I can do to fix this?

userx14, 2024-02-14:
The first issue you see is likely connected to the bluetooth library used in the project. They also have an issue with the exact same problem: https://github.com/hbldh/bleak/issues/1434 . Maybe the workaround presented in the thread also works in your case.

In the second case, you can get rid of this assertion error by commenting `#` this line:
https://github.com/userx14/omblepy/blob/1f43fefc835f5b40033b0df1d39525060bb8c096/omblepy.py#L387
The bluetooth backend I'm using, the `.unpair()` in the line above should also disconnect the device. But that does not seem to work on windows, hence the extra disconnect statement. On linux this likely works as described and the second disconnect call leads to the error you are seeing. 
You could also just ignore the error, this is last line before the program terminates anyway.

tecbeat, 2024-04-18:
I think i have a similar error on my Mac.

```
2024-04-18 15:25:48,302 - omblepy - INFO - Attempt connecting to 73FDC143-7F01-56F1-CB71-A5DCDDAB08A2.
2024-04-18 15:25:52,009 - omblepy - INFO - unpair and disconnect
Traceback (most recent call last):
  File "/Users/xxx/tmp/omblepy/./omblepy.py", line 389, in <module>
    asyncio.run(main())
  File "/opt/homebrew/Cellar/python@3.12/3.12.2_1/Frameworks/Python.framework/Versions/3.12/lib/python3.12/asyncio/runners.py", line 194, in run
    return runner.run(main)
           ^^^^^^^^^^^^^^^^
  File "/opt/homebrew/Cellar/python@3.12/3.12.2_1/Frameworks/Python.framework/Versions/3.12/lib/python3.12/asyncio/runners.py", line 118, in run
    return self._loop.run_until_complete(task)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/homebrew/Cellar/python@3.12/3.12.2_1/Frameworks/Python.framework/Versions/3.12/lib/python3.12/asyncio/base_events.py", line 685, in run_until_complete
    return future.result()
           ^^^^^^^^^^^^^^^
  File "/Users/xxx/tmp/omblepy/./omblepy.py", line 364, in main
    await bleClient.pair(protection_level = 2)
  File "/Users/xxx/tmp/omblepy/venv/lib/python3.12/site-packages/bleak/__init__.py", line 629, in pair
    return await self._backend.pair(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/xxx/tmp/omblepy/venv/lib/python3.12/site-packages/bleak/backends/corebluetooth/client.py", line 191, in pair
    raise NotImplementedError("Pairing is not available in Core Bluetooth.")
NotImplementedError: Pairing is not available in Core Bluetooth.
```

Versions:
```
Package                        Version
------------------------------ -------
bleak                          0.21.1
pip                            24.0
pyobjc-core                    9.2
pyobjc-framework-Cocoa         9.2
pyobjc-framework-CoreBluetooth 9.2
pyobjc-framework-libdispatch   9.2
terminaltables                 3.1.10
```

userx14, 2024-04-18:
Hi @tecbeat ,

I'm not so confident that these problems are related.

The bluetooth library I use does not support pairing on mac, see:
https://bleak.readthedocs.io/en/latest/backends/macos.html#bleak.backends.corebluetooth.client.BleakClientCoreBluetooth.pair

Maybe it works if you just comment out this line:
https://github.com/userx14/omblepy/blob/8297ce618c7b6877097f06dfdce9984f65733498/omblepy.py#L364

I did not have the chance to test with mac os so far.

Best,
Benjamin