userx14/omblepy #23: BP653 / HEM-6321T-Z

jamessherring, 2024-03-06:
Hi, I get exception trying to pair with Omron BP653 / HEM-6321T-Z;

On Ubuntu 22.04 I get this:
```
$ python ./omblepy.py -p -d HEM-7322T --loggerDebug
2024-03-05 16:19:23,121 - omblepy - INFO - Attempt to import module for device hem-7322t
To improve your chance of a successful connection please do the following:
 -remove previous device pairings in your OS's bluetooth dialog
 -enable bluetooth on you omron device and use the specified mode (pairing or normal)
 -do not accept any pairing dialog until you selected your device in the following list

Select your Omron device from the list below...
/media/james/data/disposable/git/my_ubpm/omblepy/./omblepy.py:297: FutureWarning: BLEDevice.rssi is deprecated and will be removed in a future version of Bleak, use AdvertisementData.rssi instead
  devices = sorted(devices, key = lambda x: x.rssi, reverse=True)
/media/james/data/disposable/git/my_ubpm/omblepy/./omblepy.py:301: FutureWarning: BLEDevice.rssi is deprecated and will be removed in a future version of Bleak, use AdvertisementData.rssi instead
  tableEntries.append([deviceIdx, device.address, device.name, device.rssi])
+----+-------------------+-------------------------------+------+
| ID | MAC               | NAME                          | RSSI |
+----+-------------------+-------------------------------+------+
| 0  | 70:E3:E4:1D:9A:A9 | 70-E3-E4-1D-9A-A9             | -76  |
| 1  | DF:46:B5:26:78:DB | DF-46-B5-26-78-DB             | -78  |
| 2  | 70:BB:6A:D6:31:6B | 70-BB-6A-D6-31-6B             | -83  |
| 3  | CF:05:1C:67:04:00 | BLEsmart_000001180F051C670400 | -85  |
| 4  | D4:9D:C0:84:03:89 | D4-9D-C0-84-03-89             | -92  |
+----+-------------------+-------------------------------+------+
Enter ID or just press Enter to rescan.
3
2024-03-05 16:19:32,733 - omblepy - INFO - Attempt connecting to CF:05:1C:67:04:00.
2024-03-05 16:20:09,851 - omblepy - INFO - unpair and disconnect
Traceback (most recent call last):
  File "/media/james/data/disposable/git/my_ubpm/omblepy/./omblepy.py", line 389, in <module>
    asyncio.run(main())
  File "/usr/lib/python3.10/asyncio/runners.py", line 44, in run
    return loop.run_until_complete(main)
  File "/usr/lib/python3.10/asyncio/base_events.py", line 649, in run_until_complete
    return future.result()
  File "/media/james/data/disposable/git/my_ubpm/omblepy/./omblepy.py", line 364, in main
    await bleClient.pair(protection_level = 2)
  File "/home/james/.local/lib/python3.10/site-packages/bleak/__init__.py", line 629, in pair
    return await self._backend.pair(*args, **kwargs)
  File "/home/james/.local/lib/python3.10/site-packages/bleak/backends/bluezdbus/client.py", line 473, in pair
    assert_reply(reply)
  File "/home/james/.local/lib/python3.10/site-packages/bleak/backends/bluezdbus/utils.py", line 22, in assert_reply
    raise BleakDBusError(reply.error_name, reply.body)
bleak.exc.BleakDBusError: [org.bluez.Error.AuthenticationCanceled] Authentication Canceled
```

On win11 I get this:
```
python ./omblepy.py -p -d HEM-7322T --loggerDebug
2024-03-03 13:55:14,756 - omblepy - INFO - Attempt to import module for device hem-7322t
To improve your chance of a successful connection please do the following:
 -remove previous device pairings in your OS's bluetooth dialog
 -enable bluetooth on you omron device and use the specified mode (pairing or normal)
 -do not accept any pairing dialog until you selected your device in the following list

Select your Omron device from the list below...
C:\data\git\temp\omblepy\omblepy.py:297: FutureWarning: BLEDevice.rssi is deprecated and will be removed in a future version of Bleak, use AdvertisementData.rssi instead
  devices = sorted(devices, key = lambda x: x.rssi, reverse=True)
C:\data\git\temp\omblepy\omblepy.py:301: FutureWarning: BLEDevice.rssi is deprecated and will be removed in a future version of Bleak, use AdvertisementData.rssi instead
  tableEntries.append([deviceIdx, device.address, device.name, device.rssi])
+----+-------------------+-------------------------------+------+
| ID | MAC               | NAME                          | RSSI |
+----+-------------------+-------------------------------+------+
| 0  | CF:05:1C:67:04:00 | BLEsmart_000001180F051C670400 | -67  |
| 1  | 43:F7:98:A5:4E:EE | None                          | -69  |
| 2  | D4:9D:C0:84:03:89 | None                          | -91  |
| 3  | 77:9B:43:C1:5F:0D | None                          | -93  |
+----+-------------------+-------------------------------+------+
Enter ID or just press Enter to rescan.
0
2024-03-03 13:55:24,974 - omblepy - INFO - Attempt connecting to CF:05:1C:67:04:00.
2024-03-03 13:55:29,683 - omblepy - INFO - unpair and disconnect
Traceback (most recent call last):
  File "C:\data\git\temp\omblepy\omblepy.py", line 389, in <module>
    asyncio.run(main())
  File "C:\Users\james\AppData\Local\Programs\Python\Python312\Lib\asyncio\runners.py", line 194, in run
    return runner.run(main)
           ^^^^^^^^^^^^^^^^
  File "C:\Users\james\AppData\Local\Programs\Python\Python312\Lib\asyncio\runners.py", line 118, in run
    return self._loop.run_until_complete(task)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\james\AppData\Local\Programs\Python\Python312\Lib\asyncio\base_events.py", line 685, in run_until_complete
    return future.result()
           ^^^^^^^^^^^^^^^
  File "C:\data\git\temp\omblepy\omblepy.py", line 364, in main
    await bleClient.pair(protection_level = 2)
  File "C:\data\git\temp\omblepy\venv\Lib\site-packages\bleak\__init__.py", line 629, in pair
    return await self._backend.pair(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\data\git\temp\omblepy\venv\Lib\site-packages\bleak\backends\winrt\client.py", line 616, in pair
    raise BleakError(f"Could not pair with device: {pairing_result.status}")
bleak.exc.BleakError: Could not pair with device: 15
```

I have tried the troubleshooting steps without success.

My BT info:
```
$ bluetoothd -v
5.64

$ hciconfig -a
hci0:	Type: Primary  Bus: USB
	BD Address: 3C:E9:F7:F3:2F:63  ACL MTU: 1021:4  SCO MTU: 96:6
	UP RUNNING 
	RX bytes:147329 acl:1133 sco:0 events:7615 errors:0
	TX bytes:843472 acl:955 sco:0 commands:4677 errors:0
	Features: 0xbf 0xfe 0x0f 0xfe 0xdb 0xff 0x7b 0x87
	Packet type: DM1 DM3 DM5 DH1 DH3 DH5 HV1 HV2 HV3 
	Link policy: RSWITCH SNIFF 
	Link mode: PERIPHERAL ACCEPT 
	Name: 'XPS13-9320-ubuntu'
	Class: 0x7c010c
	Service Classes: Rendering, Capturing, Object Transfer, Audio, Telephony
	Device Class: Computer, Laptop
	HCI Version:  (0xc)  Revision: 0x35fc
	LMP Version:  (0xc)  Subversion: 0x35fc
	Manufacturer: Intel Corp. (2)
 ```
Thanks,
James

userx14, 2024-03-07:
Hi James,

from experience getting the pairing to work can be a bit tricky, though is easyer on windows. Bluez or the bluetooth library I use (bleak) seems to have some problem with pairing on certain bluez versions. My tool only has very little control over the pairing process, this is all done in the library / os level. Here is the entirety of code related to the bluetooth pairing:
https://github.com/userx14/omblepy/blob/1f43fefc835f5b40033b0df1d39525060bb8c096/omblepy.py#L362-L364
On thing you could try is to lower the protection level to `1` or `0`.

When I trace back the error returned from bleak on windows according to the [microsoft documentation](https://learn.microsoft.com/en-us/uwp/api/windows.devices.enumeration.devicepairingresultstatus) it resolves to:

> OperationAlreadyInProgress | 15 | The device object is already attempting to pair or unpair.

Maybe you need to give windows more time for the previous connection attempt to timeout.

I would have some additional questions regarding your testing on win11:

- Did you start with the device unpaired using the bluetooth system settings for each test run?
- Do you get a pairing dialog on Windows?
- Did you accept said pairing dialog relativley quickly (<5 seconds)?

From your `hciconfig -a` I can deduce that you use a xps13. I think this model uses an intel AX211.
The intel combo cards usually work well with the omron devices, 
though I have to admit that I tested only an AX200 on windows 10.

Do you have another windows device you could test on?

Best,
Benjamin

jamessherring, 2024-03-12:
Hi Benjamin,

> Maybe you need to give windows more time for the previous connection attempt to timeout.
I cant give it any more time, it already fails :)
I respond to the pairing dialog as fast as I can, but still get the same exception.

Yes, I start with the device unpaired. After selecting my bpm in omblepy, I get the pairing pop-up in windows and "HEM-6321T-Z Connection succeeded" dialog.

I have tried variations on pairing first and without -p option, but still no success.
After pairing as above, If I re-power the BPM in connect mode (not pair mode), I get this: 
```
(venv) C:\data\git\temp\omblepy>python ./omblepy.py -d HEM-7322T
2024-03-12 00:30:57,501 - omblepy - INFO - Attempt to import module for device hem-7322t
To improve your chance of a successful connection please do the following:
 -remove previous device pairings in your OS's bluetooth dialog
 -enable bluetooth on you omron device and use the specified mode (pairing or normal)
 -do not accept any pairing dialog until you selected your device in the following list

Select your Omron device from the list below...
C:\data\git\temp\omblepy\omblepy.py:297: FutureWarning: BLEDevice.rssi is deprecated and will be removed in a future version of Bleak, use AdvertisementData.rssi instead
  devices = sorted(devices, key = lambda x: x.rssi, reverse=True)
C:\data\git\temp\omblepy\omblepy.py:301: FutureWarning: BLEDevice.rssi is deprecated and will be removed in a future version of Bleak, use AdvertisementData.rssi instead
  tableEntries.append([deviceIdx, device.address, device.name, device.rssi])
+----+-------------------+-------------------------------+------+
| ID | MAC               | NAME                          | RSSI |
+----+-------------------+-------------------------------+------+
| 0  | CF:05:1C:67:04:00 | BLEsmart_000001180F051C670400 | -59  |
| 1  | 66:74:C4:4A:6B:00 | None                          | -70  |
| 2  | CC:F0:82:0C:6C:60 | None                          | -71  |
| 3  | D4:9D:C0:84:03:89 | [TV] Samsung Q90 Series (55)  | -82  |
+----+-------------------+-------------------------------+------+
Enter ID or just press Enter to rescan.
0
2024-03-12 00:31:04,897 - omblepy - INFO - Attempt connecting to CF:05:1C:67:04:00.
2024-03-12 00:31:06,741 - omblepy - INFO - unpair and disconnect
Traceback (most recent call last):
  File "C:\data\git\temp\omblepy\omblepy.py", line 389, in <module>
    asyncio.run(main())
  File "C:\Users\james\AppData\Local\Programs\Python\Python312\Lib\asyncio\runners.py", line 194, in run
    return runner.run(main)
           ^^^^^^^^^^^^^^^^
  File "C:\Users\james\AppData\Local\Programs\Python\Python312\Lib\asyncio\runners.py", line 118, in run
    return self._loop.run_until_complete(task)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\james\AppData\Local\Programs\Python\Python312\Lib\asyncio\base_events.py", line 685, in run_until_complete
    return future.result()
           ^^^^^^^^^^^^^^^
  File "C:\data\git\temp\omblepy\omblepy.py", line 364, in main
    await bleClient.pair(protection_level = 1)
  File "C:\data\git\temp\omblepy\venv\Lib\site-packages\bleak\__init__.py", line 629, in pair
    return await self._backend.pair(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\data\git\temp\omblepy\venv\Lib\site-packages\bleak\backends\winrt\client.py", line 585, in pair
    self._requester.device_information.id
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
AttributeError: 'NoneType' object has no attribute 'device_information'
```

Unfortunately I don't have access to another windows device for now.

I am able to connect to the BPM (on ubuntu) with bettercap to get some info on device BLE attributes, log attached.
[hem6321t_bettercap_log.txt](https://github.com/userx14/omblepy/files/14565807/hem6321t_bettercap_log.txt)

Thanks,
James

userx14, 2024-03-19:
Hi James,

sorry it took so long. I have created a bootable win11 usb stick and tested on my pc with an ax200 intel bluetooth card.
My first test was with the latest python version 3.12.2, but the device scanner did not work with the error message:
```
ImportError: DLL load failed while importing _winrt_windows_devices_bluetooth: The specified module could not be found.
```

My second test was using python 3.10.11 and the most recent versions of bleak and terminaltables. 
The device scanner did work, but when I tried pairing with:
`python .\omblepy.py -d hem-7322t -p` and the device in blinking -P- mode, 
I got the same error as you:
`bleak.exc.BleakError: Could not pair with device: 15`.

After multiple attempts I got it working by instantly clicking the system pairing notification and clicking accept. The timeout for this is extremely short, if you see "unpair and disconnect" in the terminal you are already too late.
I could improve my success rate by specifying a mac address like so:
`python .\omblepy.py -d hem-7322t -p -m D3:07:19:90:29:00`. 
```
python .\omblepy.py -d hem-7322t -p -m D3:07:19:90:29:00
2024-03-19 20:29:09,917 - omblepy - INFO - Attempt to import module for device hem-7322t
2024-03-19 20:29:09,932 - omblepy - INFO - Attempt connecting to D3:07:19:90:29:00.
2024-03-19 20:29:18,136 - omblepy - INFO - Paired device successfully with new key bytearray(b'\xde\xad\xbe\xaf\x124\x124\xde\xad\xbe\xaf\x124\x124').
2024-03-19 20:29:18,184 - omblepy - INFO - From now on you can connect ommit the -p flag, even on other PCs with different bluetooth-mac-addresses.
2024-03-19 20:29:19,851 - omblepy - INFO - unpair and disconnect
```
Also having the bluetooth menu already open, decreases the time for the accept dialog to show up.

Interestingly this seems to only work every second time. The next attempt returned error 15 again, and I had to remove the device from the "Other devices" list in the Bluetooth system settings for the next attempt to work again. 

I will ask the developers of bleak if they have any idea what could cause this behavior.
Please let me know if this method works for you.

Best,
Benjamin

userx14, 2024-03-19:
FYI https://github.com/hbldh/bleak/issues/1528

jamessherring, 2024-04-12:
Hi Benjamin,

Thanks!

It's took me a number of sessions and many attempts, but encouraged by your message I persevered and eventually succeeded. 

```
(venv) C:\data\git\temp\omblepy>python .\omblepy.py -d hem-7322t -p -m CF:05:1C:67:04:00
2024-04-12 10:56:40,153 - omblepy - INFO - Attempt to import module for device hem-7322t
2024-04-12 10:56:40,232 - omblepy - INFO - Attempt connecting to CF:05:1C:67:04:00.
2024-04-12 10:56:45,765 - omblepy - INFO - Paired device successfully with new key bytearray(b'\xde\xad\xbe\xaf\x124\x124\xde\xad\xbe\xaf\x124\x124').
2024-04-12 10:56:45,765 - omblepy - INFO - From now on you can connect ommit the -p flag, even on other PCs with different bluetooth-mac-addresses.
2024-04-12 10:56:46,472 - omblepy - INFO - unpair and disconnect
```
Many thanks,
James

closed 2024-04-12 by jamessherring

userx14, 2024-04-12:
Hi James,

so the windows 11 bluetooth pairing behavior really was the problem. 
I will add a warning note in the readme, thanks for pointing out the problem. 
Unfortunately it looks like this win11 problem is not easily resolvable by the bleak developers either, since windows does not seem to expose the delay for pairing.

If you want to have a driver for reading the data of the hem-6321t, we could also discuss that. Not sure if an existing driver might work for this model or if some modifications are required.

Regards,
Benjamin

jamessherring, 2024-04-19:
Hi Benjamin,

Encouraged by success on win11, I was also able to get the pairing working on ubuntu 23.10.

I think the combination of my HEM-6321T and bluez has some issues, which I only resolved with restarting ubuntu before each try with omblepy. (Just restarting bluetooth service and clearing the bluetooth cache was not enough).

Yes, it would be awesome to have a driver to read the data from my HEM-6321T. How can I help with that?
The device characteristics for HEM-6321T are in the bettercap log attached earlier.

Thanks,
James

userx14, 2024-04-19:
Hi James,

please check out https://github.com/userx14/omblepy/tree/hem-6321t-testing for a prototype file for the device driver.
The memory adresses should be correct, but I don't know the mapping of the bits in each records.
Just try to read out your records and you should be able to compare what the measurements should be and identify ranges of bits.
They will likely be simmilar to the other devices with big endian order, so take a look at the other device drivers.
If you have any questions just let me know.

Best,
Benjamin