userx14/omblepy #24: HEM-7155T-EBK (M4) is the same as HEM-7155T (M400)

RobertWojtowicz, 2024-06-10:
Hi,

I want to purchase a blood pressure monitor and I am considering the Omron M4 Intelli model (HEM-7155T-EBK), will it work with this project ?

BR,
Robert

userx14, 2024-06-10:
Hi Robert,

as far as I can tell HEM-7155T-EBK, HEM-7155T-ESL and HEM-7155T-D
all share the same protocol / memory addresses, so I is very likely that their internals are identical. 
But I cannot give any guarante without testing and I do not own these devices.

Best,
Benjamin

RobertWojtowicz, 2024-06-11:
Hi @userx14

Thank you for the information, I occasionally ordered Omron M7, thanks to this project I will expand the functionality of my project: https://github.com/RobertWojtowicz/miscale2garmin

closed 2024-06-11 by RobertWojtowicz