userx14/omblepy #25: Problem with paring M7 Inteli

RobertWojtowicz, 2024-06-14:
Trying to pair but no luck
```
root@adm:/home/robert/omblepy# python3 omblepy.py -p -d HEM-7361T
2024-06-14 20:41:31,876 - omblepy - INFO - Attempt to import module for device hem-7361t
To improve your chance of a successful connection please do the following:
 -remove previous device pairings in your OS's bluetooth dialog
 -enable bluetooth on you omron device and use the specified mode (pairing or normal)
 -do not accept any pairing dialog until you selected your device in the following list

Select your Omron device from the list below...
/home/robert/omblepy/omblepy.py:297: FutureWarning: BLEDevice.rssi is deprecated and will be removed in a future version of Bleak, use AdvertisementData.rssi instead
  devices = sorted(devices, key = lambda x: x.rssi, reverse=True)
/home/robert/omblepy/omblepy.py:301: FutureWarning: BLEDevice.rssi is deprecated and will be removed in a future version of Bleak, use AdvertisementData.rssi instead
  tableEntries.append([deviceIdx, device.address, device.name, device.rssi])
+----+-------------------+-------------------------------+------+
| ID | MAC               | NAME                          | RSSI |
+----+-------------------+-------------------------------+------+
| 0  | 0A:31:DA:6D:C6:B5 | 0A-31-DA-6D-C6-B5             | -55  |
| 1  | 24:FC:E5:8F:CE:BF | 24-FC-E5-8F-CE-BF             | -65  |
| 2  | 14:13:0B:44:F5:FC | 14-13-0B-44-F5-FC             | -68  |
| 3  | 28:FF:B2:32:E0:19 | BLEsmart_0000025828FFB232E019 | -78  |
| 4  | 14:13:0B:C0:AC:BE | 14-13-0B-C0-AC-BE             | -80  |
| 5  | 68:27:19:1B:31:83 | D1531102                      | -81  |
| 6  | A0:92:08:5A:95:A0 | TY                            | -83  |
| 7  | 5B:A9:59:9E:8C:92 | 5B-A9-59-9E-8C-92             | -85  |
| 8  | 58:61:6A:0C:D6:0A | 58-61-6A-0C-D6-0A             | -87  |
| 9  | 42:71:ED:4F:E8:CF | 42-71-ED-4F-E8-CF             | -88  |
| 10 | FC:03:9F:59:DE:42 | FC-03-9F-59-DE-42             | -89  |
| 11 | 3E:C7:CB:01:CB:46 | 3E-C7-CB-01-CB-46             | -90  |
| 12 | 48:4A:6E:97:50:AF | [LG] webOS TV SM8600PLA       | -90  |
| 13 | 5C:B9:19:A3:0F:57 | 5C-B9-19-A3-0F-57             | -91  |
| 14 | EE:8A:9D:75:D9:03 | EE-8A-9D-75-D9-03             | -91  |
| 15 | 8C:CE:FD:75:0C:F9 | Xiaomi Band 8 Active 0CF9     | -92  |
| 16 | FC:F2:35:D8:66:52 | FC-F2-35-D8-66-52             | -92  |
| 17 | 6E:DE:E4:0B:C8:4A | 6E-DE-E4-0B-C8-4A             | -92  |
| 18 | 4F:33:BA:40:E3:A6 | 4F-33-BA-40-E3-A6             | -93  |
| 19 | 6C:68:5F:C6:41:79 | 6C-68-5F-C6-41-79             | -93  |
| 20 | 40:00:00:B4:34:E2 | 40-00-00-B4-34-E2             | -93  |
| 21 | 4B:C7:6D:D0:45:32 | 4B-C7-6D-D0-45-32             | -93  |
| 22 | CB:10:0D:3C:9A:1C | CB-10-0D-3C-9A-1C             | -93  |
| 23 | 51:8B:3F:B8:17:07 | 51-8B-3F-B8-17-07             | -94  |
| 24 | 7C:1E:B9:43:37:FD | 7C-1E-B9-43-37-FD             | -95  |
| 25 | 32:05:A6:29:AC:21 | 32-05-A6-29-AC-21             | -95  |
| 26 | 00:16:A4:6F:B2:13 | Stratos MAXO 40/0,5-12-R7     | -95  |
| 27 | 76:11:4E:41:87:84 | 76-11-4E-41-87-84             | -95  |
+----+-------------------+-------------------------------+------+
Enter ID or just press Enter to rescan.
3
2024-06-14 20:41:40,251 - omblepy - INFO - Attempt connecting to 28:FF:B2:32:E0:19.
2024-06-14 20:41:51,802 - omblepy - INFO - unpair and disconnect
Traceback (most recent call last):
  File "/usr/local/lib/python3.11/dist-packages/bleak/backends/bluezdbus/client.py", line 214, in connect
    reply = await self._bus.call(
            ^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/dbus_fast/aio/message_bus.py", line 385, in call
    await future
asyncio.exceptions.CancelledError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/robert/omblepy/omblepy.py", line 362, in main
    await bleClient.connect()
  File "/usr/local/lib/python3.11/dist-packages/bleak/__init__.py", line 615, in connect
    return await self._backend.connect(**kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/bleak/backends/bluezdbus/client.py", line 151, in connect
    async with async_timeout(timeout):
  File "/usr/lib/python3.11/asyncio/timeouts.py", line 98, in __aexit__
    raise TimeoutError
TimeoutError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/lib/python3.11/dist-packages/bleak/backends/bluezdbus/client.py", line 532, in unpair
    assert_reply(reply)
  File "/usr/local/lib/python3.11/dist-packages/bleak/backends/bluezdbus/utils.py", line 20, in assert_reply
    raise BleakDBusError(reply.error_name, reply.body)
bleak.exc.BleakDBusError: [org.bluez.Error.DoesNotExist] Does Not Exist

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/home/robert/omblepy/omblepy.py", line 389, in <module>
    asyncio.run(main())
  File "/usr/lib/python3.11/asyncio/runners.py", line 190, in run
    return runner.run(main)
           ^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/asyncio/runners.py", line 118, in run
    return self._loop.run_until_complete(task)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/asyncio/base_events.py", line 653, in run_until_complete
    return future.result()
           ^^^^^^^^^^^^^^^
  File "/home/robert/omblepy/omblepy.py", line 386, in main
    await bleClient.unpair()
  File "/usr/local/lib/python3.11/dist-packages/bleak/__init__.py", line 653, in unpair
    return await self._backend.unpair()
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/bleak/backends/bluezdbus/client.py", line 535, in unpair
    raise BleakDeviceNotFoundError(
bleak.exc.BleakDeviceNotFoundError: Device with address 28:FF:B2:32:E0:19 was not found.
```

userx14, 2024-06-14:
Hi Robert,

this is most likely something wrong in the bluetooth library or on os level with the bluetooth connection and pairing. 
I assume this is a linux machine? Do you get a pairing confirmation dialog in your os bluetooth dialog?

From my testing only very specific sets of bluez, bluetooth adapters and linux kernel versions did work.
What bluetooth adapter are you using?

If you have the possibility please test with win10 to rule out the operating system as the cause, 
as that was the most reproducible setup unfortunately. 
The windows 10 live boot images did also work in my testing.

Best
Benjamin

RobertWojtowicz, 2024-06-14:
Hi @userx14

I am using Debian 12, I am using ssh console, for this you need a graphical mode ?
I use usb bluetooth 5.3 on ATS2851 chipset (supported natively by Debian 12). It works fine for me when syncing Xiaomi scales.
Unfortunately I do not have Windows 10 I have Windows 11, but the solution wants to use with Linux.

userx14, 2024-06-14:
Hi Robert,

If I remember correctely, you could use bluetoothctl on the terminal, but you need to do this simulteneously to running the python script with something like a terminal multiplexer. 

The Camebridge Silicon Stick will work on linux, but NOT on Windows. ATS2851 sounds fine for both OS.
Windows 11 is sadly not so simple for testing, see: https://github.com/userx14/omblepy/issues/23#issuecomment-2008148253

Regards,
Benjamin

RobertWojtowicz, 2024-06-15:
Unfortunately, it doesn't want to pair:
```
^Croot@adm:~# bluetoothctl pair 28:FF:B2:32:E0:19
Attempting to pair with 28:FF:B2:32:E0:19
Failed to pair: org.bluez.Error.AuthenticationFailed
```

RobertWojtowicz, 2024-06-16:
@userx14 

Is it possible to read data without pairing ?

BR,
Robert

RobertWojtowicz, 2024-06-16:
Now your script returns this:
```
root@adm:/home/robert/omblepy# python3 omblepy.py -p -d HEM-7361T --loggerDebug
2024-06-16 12:34:16,154 - omblepy - INFO - Attempt to import module for device hem-7361t
To improve your chance of a successful connection please do the following:
 -remove previous device pairings in your OS's bluetooth dialog
 -enable bluetooth on you omron device and use the specified mode (pairing or normal)
 -do not accept any pairing dialog until you selected your device in the following list

Select your Omron device from the list below...
/home/robert/omblepy/omblepy.py:297: FutureWarning: BLEDevice.rssi is deprecated and will be removed in a future version of Bleak, use AdvertisementData.rssi instead
  devices = sorted(devices, key = lambda x: x.rssi, reverse=True)
/home/robert/omblepy/omblepy.py:301: FutureWarning: BLEDevice.rssi is deprecated and will be removed in a future version of Bleak, use AdvertisementData.rssi instead
  tableEntries.append([deviceIdx, device.address, device.name, device.rssi])
+----+-------------------+-------------------------------+------+
| ID | MAC               | NAME                          | RSSI |
+----+-------------------+-------------------------------+------+
| 0  | 24:FC:E5:8F:CE:BF | 24-FC-E5-8F-CE-BF             | -66  |
| 1  | 10:0E:68:2C:6A:9F | 10-0E-68-2C-6A-9F             | -70  |
| 2  | A0:92:08:5A:95:A0 | TY                            | -76  |
| 3  | 54:A1:FC:06:4A:6A | 54-A1-FC-06-4A-6A             | -81  |
| 4  | 14:13:0B:44:F5:FC | 14-13-0B-44-F5-FC             | -86  |
| 5  | 28:FF:B2:32:E0:19 | BLEsmart_0000025828FFB232E019 | -90  |
| 6  | 68:27:19:1B:31:83 | D1531102                      | -93  |
+----+-------------------+-------------------------------+------+
Enter ID or just press Enter to rescan.
5
2024-06-16 12:34:42,741 - omblepy - INFO - Attempt connecting to 28:FF:B2:32:E0:19.
2024-06-16 12:35:00,542 - omblepy - INFO - unpair and disconnect
Traceback (most recent call last):
  File "/usr/local/lib/python3.11/dist-packages/bleak/backends/bluezdbus/client.py", line 273, in connect
    await self.get_services(
  File "/usr/local/lib/python3.11/dist-packages/bleak/backends/bluezdbus/client.py", line 661, in get_services
    self.services = await manager.get_services(
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/bleak/backends/bluezdbus/manager.py", line 666, in get_services
    await self._wait_for_services_discovery(device_path)
  File "/usr/local/lib/python3.11/dist-packages/bleak/backends/bluezdbus/manager.py", line 783, in _wait_for_services_discovery
    done, _ = await asyncio.wait(
              ^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/asyncio/tasks.py", line 418, in wait
    return await _wait(fs, timeout, return_when, loop)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/asyncio/tasks.py", line 525, in _wait
    await waiter
asyncio.exceptions.CancelledError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/robert/omblepy/omblepy.py", line 389, in <module>
    asyncio.run(main())
  File "/usr/lib/python3.11/asyncio/runners.py", line 190, in run
    return runner.run(main)
           ^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/asyncio/runners.py", line 118, in run
    return self._loop.run_until_complete(task)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/asyncio/base_events.py", line 653, in run_until_complete
    return future.result()
           ^^^^^^^^^^^^^^^
  File "/home/robert/omblepy/omblepy.py", line 362, in main
    await bleClient.connect()
  File "/usr/local/lib/python3.11/dist-packages/bleak/__init__.py", line 615, in connect
    return await self._backend.connect(**kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/bleak/backends/bluezdbus/client.py", line 151, in connect
    async with async_timeout(timeout):
  File "/usr/lib/python3.11/asyncio/timeouts.py", line 98, in __aexit__
    raise TimeoutError
TimeoutError
```

userx14, 2024-06-16:
Hi Robert,

> Which version of bluez are you using ?

I think it was an old ubuntu version with bluez 5.55. 
But that was some time ago, so I don't remember all the details.
For most of my testing I used Win 10.

You could also try [ubpm](https://codeberg.org/LazyT/ubpm), he uses the QT Bluetooth library. 
But ubpm requires cloning the bluetooth mac address from a phone with the omron app, 
this should work with your csr stick though.

> Is it possible to read data without pairing ?

You need bluetooth pairing to work, in order to read out the data. 
Without pairing it returns error codes when you try to write / read to the unlock characteristic.

> Now your script returns this: [...]

This still looks like some error in the bluetooth library (_wait_for_services_discovery).

Regards,
Benjamin

RobertWojtowicz, 2024-06-16:
@userx14  Thanks for your reply,
A little progress, Raspberry PI0W with integrated BT and Debian 12 32-bit (Linux raspberrypi 6.6.31+rpt-rpi-v6):
Pairing works
```
pi@raspberrypi:~/omblepy $ python3 omblepy.py -p -d HEM-7361T
2024-06-16 13:49:51,005 - omblepy - INFO - Attempt to import module for device hem-7361t
To improve your chance of a successful connection please do the following:
 -remove previous device pairings in your OS's bluetooth dialog
 -enable bluetooth on you omron device and use the specified mode (pairing or normal)
 -do not accept any pairing dialog until you selected your device in the following list

Select your Omron device from the list below...
/home/pi/omblepy/omblepy.py:297: FutureWarning: BLEDevice.rssi is deprecated and will be removed in a future version of Bleak, use AdvertisementData.rssi instead
  devices = sorted(devices, key = lambda x: x.rssi, reverse=True)
/home/pi/omblepy/omblepy.py:301: FutureWarning: BLEDevice.rssi is deprecated and will be removed in a future version of Bleak, use AdvertisementData.rssi instead
  tableEntries.append([deviceIdx, device.address, device.name, device.rssi])
+----+-------------------+-------------------------------+------+
| ID | MAC               | NAME                          | RSSI |
+----+-------------------+-------------------------------+------+
| 0  | 0A:F7:B6:36:9B:19 | 0A-F7-B6-36-9B-19             | -43  |
| 1  | 28:FF:B2:32:E0:19 | BLEsmart_0000025828FFB232E019 | -47  |
| 2  | 14:13:0B:44:F5:FC | 14-13-0B-44-F5-FC             | -50  |
| 3  | 24:FC:E5:8F:CE:BF | 24-FC-E5-8F-CE-BF             | -56  |
| 4  | 75:E4:18:07:80:DB | 75-E4-18-07-80-DB             | -75  |
| 5  | 68:27:A8:E5:EF:15 | 68-27-A8-E5-EF-15             | -76  |
| 6  | 6C:6F:2F:F2:49:A3 | 6C-6F-2F-F2-49-A3             | -83  |
| 7  | D5:F0:73:F7:0C:F0 | D5-F0-73-F7-0C-F0             | -84  |
| 8  | 65:B4:EB:9D:FF:87 | 65-B4-EB-9D-FF-87             | -94  |
| 9  | A0:92:08:5A:95:A0 | TY                            | -95  |
| 10 | 38:01:95:9B:80:D1 | 38-01-95-9B-80-D1             | -95  |
| 11 | 7D:33:F0:58:C0:38 | 7D-33-F0-58-C0-38             | -97  |
| 12 | 68:27:19:1B:31:83 | D1531102                      | -97  |
+----+-------------------+-------------------------------+------+
Enter ID or just press Enter to rescan.
1
2024-06-16 13:49:59,737 - omblepy - INFO - Attempt connecting to 28:FF:B2:32:E0:19.
2024-06-16 13:50:04,954 - omblepy - INFO - Paired device successfully with new key bytearray(b'\xde\xad\xbe\xaf\x124\x124\xde\xad\xbe\xaf\x124\x124').
2024-06-16 13:50:04,960 - omblepy - INFO - From now on you can connect ommit the -p flag, even on other PCs with different bluetooth-mac-addresses.
2024-06-16 13:50:05,629 - omblepy - INFO - unpair and disconnect
Traceback (most recent call last):
  File "/home/pi/omblepy/omblepy.py", line 389, in <module>
    asyncio.run(main())
  File "/usr/lib/python3.11/asyncio/runners.py", line 190, in run
    return runner.run(main)
           ^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/asyncio/runners.py", line 118, in run
    return self._loop.run_until_complete(task)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/asyncio/base_events.py", line 653, in run_until_complete
    return future.result()
           ^^^^^^^^^^^^^^^
  File "/home/pi/omblepy/omblepy.py", line 387, in main
    await bleClient.disconnect()
  File "/usr/local/lib/python3.11/dist-packages/bleak/__init__.py", line 624, in disconnect
    return await self._backend.disconnect()
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/bleak/backends/bluezdbus/client.py", line 425, in disconnect
    assert self._bus is None
           ^^^^^^^^^^^^^^^^^
AssertionError
```
Download data works:
```
pi@raspberrypi:~/omblepy $ python3 omblepy.py -d HEM-7361T --mac 28:FF:B2:32:E0:19
2024-06-16 14:14:59,961 - omblepy - INFO - Attempt to import module for device hem-7361t
2024-06-16 14:15:00,442 - omblepy - INFO - Attempt connecting to 28:FF:B2:32:E0:19.
2024-06-16 14:15:05,845 - omblepy - INFO - communication started
2024-06-16 14:15:06,510 - omblepy - INFO - start reading data, this can take a while, use debug flag to see progress
2024-06-16 14:15:36,948 - omblepy - INFO - communication finished
2024-06-16 14:15:37,058 - omblepy - INFO - writing data to user1.csv
2024-06-16 14:15:37,141 - omblepy - INFO - writing data to user2.csv
2024-06-16 14:15:37,370 - omblepy - INFO - unpair and disconnect
Traceback (most recent call last):
  File "/home/pi/omblepy/omblepy.py", line 389, in <module>
    asyncio.run(main())
  File "/usr/lib/python3.11/asyncio/runners.py", line 190, in run
    return runner.run(main)
           ^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/asyncio/runners.py", line 118, in run
    return self._loop.run_until_complete(task)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/asyncio/base_events.py", line 653, in run_until_complete
    return future.result()
           ^^^^^^^^^^^^^^^
  File "/home/pi/omblepy/omblepy.py", line 387, in main
    await bleClient.disconnect()
  File "/usr/local/lib/python3.11/dist-packages/bleak/__init__.py", line 624, in disconnect
    return await self._backend.disconnect()
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/bleak/backends/bluezdbus/client.py", line 425, in disconnect
    assert self._bus is None
           ^^^^^^^^^^^^^^^^^
AssertionError
```

Now I need to figure out why it doesn't work on x86 virtual machine with BT 4.0 (and 5.3) with Debian 12 64bit, maybe some ideas ?

BR,
Robert

RobertWojtowicz, 2024-06-16:
@userx14 

I made a clone of the MAC address of the paired BT 4.1 Raspberry PI and changed the MAC to BT 4.0 on which the omblepy reading does not work.
Unfortunately this did not help, I would like a simple way from the command line to read the measurement data.

BR,
Robert

userx14, 2024-06-16:
I'm not really sure what it could be, but it is interesting that you got it working with a pi zero w.
Maybe it is related to the differences in the bluetooth driver of the broadcom chipset, 
or it might be some configuration regarding bluetooth le, like encryption or the device supported pairing modes. 
Did you try changing them with `btmgmt`?

Regards
Benjamin

RobertWojtowicz, 2024-06-17:
HI @userx14 

I came to the same thoughts, that it is the fault of BT firmware or drivers. I will continue to fight with BT :)

BR,
Robert

RobertWojtowicz, 2024-06-17:
Same problem with BT5.0/5.1 on RTL8761B chipset.

RobertWojtowicz, 2024-06-17:
Hi, @userx14 

I got a good scan just not sure what else helped :)

BR,
Robert

RobertWojtowicz, 2024-06-17:
https://github.com/hbldh/bleak/pull/1100
https://github.com/hbldh/bleak/issues/1434
https://github.com/hbldh/bleak/issues/758

userx14, 2024-06-17:
> Bingo! works with BT4.1 (Cambridge Silicon Radio) i BT5.1 (RTL8761B) - If bluetoothctl console is enabled then everything works, just how to solve it permanently?

Not sure how a permanent fix looks like, 
but that is what I meant with terminal multiplexer and bluetoothctl simultaneously, in my previous comment.

> It is necessary to remember that the BT of the phone with the app should be turned off because the Omron disappears from the possibility of reading data when the app connects to the Omron.
 
Probably the omron does not support multiple connections, sounds as expected.

RobertWojtowicz, 2024-06-18:
@userx14 

The workaround for the problem is very simple to the omblepy.py script I added: 
```
import subprocess
import atexit
process = subprocess.Popen(['bluetoothctl'], stdout=subprocess.DEVNULL, stderr=subprocess.DEVNULL)
atexit.register(process.terminate)
```

userx14, 2024-06-18:
Hi Robert,

thanks for posting the bash script, I think I should add a note about running bluetoothctl to the readme of this project.

> Omron sends data only 2 times data

Does that mean that only two records are sent, or that after the second time reading all the records, the transmission does not work anymore? And what part stops working, the pairing process, the unlocking process or the data readout?

Regards,
Benjamin

RobertWojtowicz, 2024-06-19:
This is probably the nature of Omron, when it connects to a paired device, it sends all the data, and then it disconnects and disappears from the search?

This is not cool when you have a paired phone and omblepy, because the phone will establish a connection faster and even when it downloads the data it doesn't have a chance to download through omblepy.

BR,
Robert

userx14, 2024-06-19:
I think it does poll the omron from time to time so that it can automatically synchronize new data. Can your disable this in the omron app options (auto synchronization)?

But to be fair, omson very likely didn't plan for third party readout tools.

Regards,
Benjamin

closed 2024-07-08 by RobertWojtowicz