userx14/omblepy #27: Support for BP786N (series 10)

maxromanovsky, 2024-07-12:
I think BP786N (series 10) has been discontinued by Omron, but it still works well and supported by the Omron Connect app.

I can't find a landing page for it anymore, but the manual is still available: https://omronhealthcare.com/wp-content/uploads/BP786N-IM_EN.pdf

I would be happy to assist with the troubleshooting :) 

I have been trying to connect to it using
- RPI Zero 2
- an external rtl8761bu BLE 5.3 dongle
- built-in Bluetooth adapter has been disabled

`bluetoothctl` running in another terminal, and when asked interactively `[agent] Accept pairing (yes/no): ` I answered `yes`

```console
$ python3 ./omblepy.py -p -d HEM-7322T -m D0:05:06:80:89:00
2024-07-12 20:22:08,578 - omblepy - INFO - Attempt to import module for device hem-7322t
2024-07-12 20:22:08,730 - omblepy - INFO - Attempt connecting to D0:05:06:80:89:00.
2024-07-12 20:22:22,578 - omblepy - INFO - unpair and disconnect
Traceback (most recent call last):
  File "/home/max/omblepy-main/./omblepy.py", line 367, in main
    raise OSError("""Some required bluetooth attributes not found on this ble device.
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
OSError: Some required bluetooth attributes not found on this ble device.
                             This means that either, you connected to a wrong device,
                             or that your OS has a bug when reading BT LE device attributes (certain linux versions).

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/max/omblepy-main/./omblepy.py", line 389, in <module>
    asyncio.run(main())
  File "/usr/lib/python3.11/asyncio/runners.py", line 190, in run
    return runner.run(main)
           ^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/asyncio/runners.py", line 118, in run
    return self._loop.run_until_complete(task)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/asyncio/base_events.py", line 653, in run_until_complete
    return future.result()
           ^^^^^^^^^^^^^^^
  File "/home/max/omblepy-main/./omblepy.py", line 387, in main
    await bleClient.disconnect()
  File "/usr/local/lib/python3.11/dist-packages/bleak/__init__.py", line 624, in disconnect
    return await self._backend.disconnect()
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/bleak/backends/bluezdbus/client.py", line 425, in disconnect
    assert self._bus is None
           ^^^^^^^^^^^^^^^^^
AssertionError
```

`bluetoothctl` output:
```
[...]
[CHG] Device D0:05:06:80:89:00 Connected: yes
Request authorization
[agent] Accept pairing (yes/no): [NEW] Secondary Service (Handle 0x0000)
    /org/bluez/hci0/dev_D0_05_06_80_89_00/service0100
    00001801-0000-1000-8000-00805f9b34fb
    Generic Attribute Profile
[agent] Accept pairing (yes/no): [NEW] Characteristic (Handle 0x0000)
    /org/bluez/hci0/dev_D0_05_06_80_89_00/service0100/char0110
    00002a05-0000-1000-8000-00805f9b34fb
    Service Changed
[agent] Accept pairing (yes/no): [NEW] Descriptor (Handle 0x0000)
    /org/bluez/hci0/dev_D0_05_06_80_89_00/service0100/char0110/desc0112
    00002902-0000-1000-8000-00805f9b34fb
    Client Characteristic Configuration
[agent] Accept pairing (yes/no): [NEW] Secondary Service (Handle 0x0000)
    /org/bluez/hci0/dev_D0_05_06_80_89_00/service0200
    0000180a-0000-1000-8000-00805f9b34fb
    Device Information
[agent] Accept pairing (yes/no): [NEW] Characteristic (Handle 0x0000)
    /org/bluez/hci0/dev_D0_05_06_80_89_00/service0200/char0210
    00002a23-0000-1000-8000-00805f9b34fb
    System ID
[agent] Accept pairing (yes/no): [NEW] Characteristic (Handle 0x0000)
    /org/bluez/hci0/dev_D0_05_06_80_89_00/service0200/char0220
    00002a24-0000-1000-8000-00805f9b34fb
    Model Number String
[agent] Accept pairing (yes/no): [NEW] Characteristic (Handle 0x0000)
    /org/bluez/hci0/dev_D0_05_06_80_89_00/service0200/char0230
    00002a25-0000-1000-8000-00805f9b34fb
    Serial Number String
[agent] Accept pairing (yes/no): [NEW] Characteristic (Handle 0x0000)
    /org/bluez/hci0/dev_D0_05_06_80_89_00/service0200/char0240
    00002a26-0000-1000-8000-00805f9b34fb
    Firmware Revision String
[agent] Accept pairing (yes/no): [NEW] Characteristic (Handle 0x0000)
    /org/bluez/hci0/dev_D0_05_06_80_89_00/service0200/char0250
    00002a27-0000-1000-8000-00805f9b34fb
    Hardware Revision String
[agent] Accept pairing (yes/no): [NEW] Characteristic (Handle 0x0000)
    /org/bluez/hci0/dev_D0_05_06_80_89_00/service0200/char0260
    00002a28-0000-1000-8000-00805f9b34fb
    Software Revision String
[agent] Accept pairing (yes/no): [NEW] Characteristic (Handle 0x0000)
    /org/bluez/hci0/dev_D0_05_06_80_89_00/service0200/char0270
    00002a29-0000-1000-8000-00805f9b34fb
    Manufacturer Name String
[agent] Accept pairing (yes/no): [NEW] Characteristic (Handle 0x0000)
    /org/bluez/hci0/dev_D0_05_06_80_89_00/service0200/char0280
    00002a2a-0000-1000-8000-00805f9b34fb
    IEEE 11073-20601 Regulatory Cert. Data List
[agent] Accept pairing (yes/no): [CHG] Device D0:05:06:80:89:00 UUIDs: 00001800-0000-1000-8000-00805f9b34fb
[agent] Accept pairing (yes/no): [CHG] Device D0:05:06:80:89:00 UUIDs: 00001801-0000-1000-8000-00805f9b34fb
[agent] Accept pairing (yes/no): [CHG] Device D0:05:06:80:89:00 UUIDs: 0000180a-0000-1000-8000-00805f9b34fb
[agent] Accept pairing (yes/no): [CHG] Device D0:05:06:80:89:00 ServicesResolved: yes
[agent] Accept pairing (yes/no): [CHG] Device D0:05:06:80:89:00 Name: BP786N
[agent] Accept pairing (yes/no): [CHG] Device D0:05:06:80:89:00 Alias: BP786N
[agent] Accept pairing (yes/no): [CHG] Device D0:05:06:80:89:00 Trusted: yes
[agent] Accept pairing (yes/no): yes
[CHG] Device D0:05:06:80:89:00 Bonded: yes
[CHG] Device D0:05:06:80:89:00 Paired: yes
[CHG] Device D0:05:06:80:89:00 ServicesResolved: no
[CHG] Device D0:05:06:80:89:00 Connected: no
[DEL] Descriptor (Handle 0x0000)
    /org/bluez/hci0/dev_D0_05_06_80_89_00/service0100/char0110/desc0112
    00002902-0000-1000-8000-00805f9b34fb
    Client Characteristic Configuration
[DEL] Characteristic (Handle 0x0000)
    /org/bluez/hci0/dev_D0_05_06_80_89_00/service0100/char0110
    00002a05-0000-1000-8000-00805f9b34fb
    Service Changed
[DEL] Secondary Service (Handle 0x0000)
    /org/bluez/hci0/dev_D0_05_06_80_89_00/service0100
    00001801-0000-1000-8000-00805f9b34fb
    Generic Attribute Profile
[DEL] Characteristic (Handle 0x0000)
    /org/bluez/hci0/dev_D0_05_06_80_89_00/service0200/char0210
    00002a23-0000-1000-8000-00805f9b34fb
    System ID
[DEL] Characteristic (Handle 0x0000)
    /org/bluez/hci0/dev_D0_05_06_80_89_00/service0200/char0220
    00002a24-0000-1000-8000-00805f9b34fb
    Model Number String
[DEL] Characteristic (Handle 0x0000)
    /org/bluez/hci0/dev_D0_05_06_80_89_00/service0200/char0230
    00002a25-0000-1000-8000-00805f9b34fb
    Serial Number String
[DEL] Characteristic (Handle 0x0000)
    /org/bluez/hci0/dev_D0_05_06_80_89_00/service0200/char0240
    00002a26-0000-1000-8000-00805f9b34fb
    Firmware Revision String
[DEL] Characteristic (Handle 0x0000)
    /org/bluez/hci0/dev_D0_05_06_80_89_00/service0200/char0250
    00002a27-0000-1000-8000-00805f9b34fb
    Hardware Revision String
[DEL] Characteristic (Handle 0x0000)
    /org/bluez/hci0/dev_D0_05_06_80_89_00/service0200/char0260
    00002a28-0000-1000-8000-00805f9b34fb
    Software Revision String
[DEL] Characteristic (Handle 0x0000)
    /org/bluez/hci0/dev_D0_05_06_80_89_00/service0200/char0270
    00002a29-0000-1000-8000-00805f9b34fb
    Manufacturer Name String
[DEL] Characteristic (Handle 0x0000)
    /org/bluez/hci0/dev_D0_05_06_80_89_00/service0200/char0280
    00002a2a-0000-1000-8000-00805f9b34fb
    IEEE 11073-20601 Regulatory Cert. Data List
[DEL] Secondary Service (Handle 0x0000)
    /org/bluez/hci0/dev_D0_05_06_80_89_00/service0200
    0000180a-0000-1000-8000-00805f9b34fb
    Device Information
[DEL] Device D0:05:06:80:89:00 BP786N
[...]
```

OS Details:

```console
$ dmesg | grep tooth
[   12.674245] Bluetooth: hci0: RTL: examining hci_ver=0a hci_rev=000b lmp_ver=0a lmp_subver=8761
[   12.690214] Bluetooth: hci0: RTL: rom_version status=0 version=1
[   12.690260] Bluetooth: hci0: RTL: loading rtl_bt/rtl8761bu_fw.bin
[   12.697960] Bluetooth: hci0: RTL: loading rtl_bt/rtl8761bu_config.bin
[   12.709476] Bluetooth: hci0: RTL: cfg_sz 6, total sz 30210
[   14.642482] Bluetooth: hci0: RTL: fw version 0xdfc6d922

$ hcitool dev
Devices:
	hci0	8A:88:4B:E0:A5:72

$ uname -a
Linux rpi02w 6.6.31+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.31-1+rpt1 (2024-05-29) aarch64 GNU/Linux

$ cat /etc/*release
PRETTY_NAME="Debian GNU/Linux 12 (bookworm)"
NAME="Debian GNU/Linux"
VERSION_ID="12"
VERSION="12 (bookworm)"
VERSION_CODENAME=bookworm
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"
```

userx14, 2024-07-12:
Hi @maxromanovsky 

the BP786N is also called HEM-7321T. 
From what I can find out it should be usable with the HEM-7322T driver.
The error is caused because omblepy does not find the bluetooth characteristic with the UUID 
`ecbe3980-c9a2-11e1-b1bd-0002a5d5c51b`.

The most reliable way to check if it exist on your omron device, is to use nRF Connect on a phone. 
When the device is paired, check if the service uuid (ecbe3980...) exists. 
Most likely it will be there, and this is a linux / bluez issue.

Best,
Benjamin

maxromanovsky, 2024-07-13:
@userx14 thank you for quick reply!

I have tried it again with the external and built-in Bluetooth on RPI Zero 2.
I couldn't figure out how to spoof the external Realtek's MAC address, so I reverted to the internal and set MAC address to be the same as my iPhone's (where the app is working).
I disabled iPhone's bluetooth just in case (as in, fully in the settings).
Still no luck, but the error is different (tried in pairing mode and in normal sync mode).
I hope to verify it on Windows later (realistically in few months).

```console
max@rpi02w:~/omblepy-main $ python3 ./omblepy.py -d HEM-7322T -m D0:05:06:80:89:00
2024-07-13 12:19:11,973 - omblepy - INFO - Attempt to import module for device hem-7322t
2024-07-13 12:19:12,120 - omblepy - INFO - Attempt connecting to D0:05:06:80:89:00.
2024-07-13 12:19:14,044 - omblepy - INFO - unpair and disconnect
Traceback (most recent call last):
  File "/home/max/omblepy-main/./omblepy.py", line 389, in <module>
    asyncio.run(main())
  File "/usr/lib/python3.11/asyncio/runners.py", line 190, in run
    return runner.run(main)
           ^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/asyncio/runners.py", line 118, in run
    return self._loop.run_until_complete(task)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/asyncio/base_events.py", line 653, in run_until_complete
    return future.result()
           ^^^^^^^^^^^^^^^
  File "/home/max/omblepy-main/./omblepy.py", line 362, in main
    await bleClient.connect()
  File "/usr/local/lib/python3.11/dist-packages/bleak/__init__.py", line 615, in connect
    return await self._backend.connect(**kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/bleak/backends/bluezdbus/client.py", line 254, in connect
    assert_reply(reply)
  File "/usr/local/lib/python3.11/dist-packages/bleak/backends/bluezdbus/utils.py", line 20, in assert_reply
    raise BleakDBusError(reply.error_name, reply.body)
bleak.exc.BleakDBusError: [org.bluez.Error.Failed] Input/output error
max@rpi02w:~/omblepy-main $ python3 ./omblepy.py -p -d HEM-7322T -m D0:05:06:80:89:00
2024-07-13 12:19:16,467 - omblepy - INFO - Attempt to import module for device hem-7322t
2024-07-13 12:19:16,641 - omblepy - INFO - Attempt connecting to D0:05:06:80:89:00.
2024-07-13 12:19:18,570 - omblepy - INFO - unpair and disconnect
Traceback (most recent call last):
  File "/home/max/omblepy-main/./omblepy.py", line 389, in <module>
    asyncio.run(main())
  File "/usr/lib/python3.11/asyncio/runners.py", line 190, in run
    return runner.run(main)
           ^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/asyncio/runners.py", line 118, in run
    return self._loop.run_until_complete(task)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/asyncio/base_events.py", line 653, in run_until_complete
    return future.result()
           ^^^^^^^^^^^^^^^
  File "/home/max/omblepy-main/./omblepy.py", line 362, in main
    await bleClient.connect()
  File "/usr/local/lib/python3.11/dist-packages/bleak/__init__.py", line 615, in connect
    return await self._backend.connect(**kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/bleak/backends/bluezdbus/client.py", line 254, in connect
    assert_reply(reply)
  File "/usr/local/lib/python3.11/dist-packages/bleak/backends/bluezdbus/utils.py", line 20, in assert_reply
    raise BleakDBusError(reply.error_name, reply.body)
bleak.exc.BleakDBusError: [org.bluez.Error.Failed] Input/output error
```

userx14, 2024-07-13:
@maxromanovsky, thanks for verifying with the nRF app. 
So looks like the device has the service, but it is not visible from your linux device.
Do you see a pairing confirmation request on your linux device? 
You can use `bluetoothctl` and `menu gatt/list-attributes` to check the visiblity of the attributes on your test device.
If thery are missing there it is very likely some linux bluetooth wierdness going on.

> `[org.bluez.Error.Failed] Input/output error`

The new error looks like something is going wrong in the bluez linux driver.

> I couldn't figure out how to spoof the external Realtek's MAC address, so I reverted to the internal and set MAC address to be the same as my iPhone's (where the app is working).

Omblepy unlike ubpm does not require BT-MAC address spoofing. When you pair one time with omblepy, it overwrites a pairing key in one bluetooth attribute. This unlocking reading the data from every pc from this point onward, until you pair with your phone again.

> I disabled iPhone's bluetooth just in case (as in, fully in the settings).

That's a good step. 
I had a report where this caused problem, because the omron app tried to autoconnect, interfering with the pairing.

maxromanovsky, 2024-07-15:
@userx14 sorry for the delayed response.

Here's how it looks like if I launch `bluetoothctl` in another terminal and respond `yes` to pairing.
<details>
<summary>omblepy</summary>

```console
$ python3 ./omblepy.py -p -d HEM-7322T -m D0:05:06:80:89:00
2024-07-15 19:55:11,886 - omblepy - INFO - Attempt to import module for device hem-7322t
2024-07-15 19:55:12,032 - omblepy - INFO - Attempt connecting to D0:05:06:80:89:00.
2024-07-15 19:55:21,115 - omblepy - INFO - unpair and disconnect
Traceback (most recent call last):
  File "/home/max/omblepy-main/./omblepy.py", line 367, in main
    raise OSError("""Some required bluetooth attributes not found on this ble device.
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
OSError: Some required bluetooth attributes not found on this ble device.
                             This means that either, you connected to a wrong device,
                             or that your OS has a bug when reading BT LE device attributes (certain linux versions).

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/max/omblepy-main/./omblepy.py", line 389, in <module>
    asyncio.run(main())
  File "/usr/lib/python3.11/asyncio/runners.py", line 190, in run
    return runner.run(main)
           ^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/asyncio/runners.py", line 118, in run
    return self._loop.run_until_complete(task)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/asyncio/base_events.py", line 653, in run_until_complete
    return future.result()
           ^^^^^^^^^^^^^^^
  File "/home/max/omblepy-main/./omblepy.py", line 387, in main
    await bleClient.disconnect()
  File "/usr/local/lib/python3.11/dist-packages/bleak/__init__.py", line 624, in disconnect
    return await self._backend.disconnect()
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/bleak/backends/bluezdbus/client.py", line 425, in disconnect
    assert self._bus is None
           ^^^^^^^^^^^^^^^^^
AssertionError
```

</details>

<details>
<summary>bluetoothctl</summary>

```
CHG] Controller B8:27:EB:E0:B4:2D Discovering: yes
[CHG] Device 62:4E:A9:46:7F:71 RSSI: -77
[CHG] Device 62:4E:A9:46:7F:71 TxPower: 8
[NEW] Device A4:C1:38:43:A2:53 LYWSD03MMC
[CHG] Device FC:A8:9B:AE:14:19 RSSI: -84
[CHG] Device 1C:AF:4A:34:00:F3 RSSI: -88
[CHG] Device 4C:F6:60:10:45:08 RSSI: -68
[CHG] Device 4C:F6:60:10:45:08 TxPower: 12
[CHG] Device 41:5B:6C:A9:32:74 RSSI: -74
[CHG] Device 45:78:11:B4:11:01 RSSI: -62
[CHG] Device 42:78:69:F9:71:90 RSSI: -58
[NEW] Device 5B:8C:73:77:5B:2E 5B-8C-73-77-5B-2E
[NEW] Device CC:19:3B:46:B5:F4 CC-19-3B-46-B5-F4
[CHG] Device 57:9F:61:88:9F:25 RSSI: -82
[CHG] Device 57:9F:61:88:9F:25 TxPower: 8
[NEW] Device D0:05:06:80:89:00 BLEsmart_0000011B100506808900
[CHG] Device 5E:F5:F9:7C:6E:D7 RSSI: -91
[CHG] Device CC:19:3B:46:B5:F4 RSSI is nil
[DEL] Device CC:19:3B:46:B5:F4 CC-19-3B-46-B5-F4
[CHG] Device 5E:F5:F9:7C:6E:D7 RSSI is nil
[... 124 lines cut ...]
	IEEE 11073-20601 Regulatory Cert. Data List
[DEL] Secondary Service (Handle 0x0000)
	/org/bluez/hci0/dev_D0_05_06_80_89_00/service0200
	0000180a-0000-1000-8000-00805f9b34fb
	Device Information
[DEL] Device D0:05:06:80:89:00 BP786N
[DEL] Device 62:4E:A9:46:7F:71 62-4E-A9-46-7F-71
[DEL] Device A4:C1:38:43:A2:53 LYWSD03MMC
[DEL] Device FC:A8:9B:AE:14:19 FC-A8-9B-AE-14-19
[DEL] Device 1C:AF:4A:34:00:F3 1C-AF-4A-34-00-F3
[DEL] Device 4C:F6:60:10:45:08 4C-F6-60-10-45-08
[DEL] Device 41:5B:6C:A9:32:74 41-5B-6C-A9-32-74
[DEL] Device 45:78:11:B4:11:01 45-78-11-B4-11-01
[DEL] Device 42:78:69:F9:71:90 42-78-69-F9-71-90
[DEL] Device 5B:8C:73:77:5B:2E 5B-8C-73-77-5B-2E
[DEL] Device 57:9F:61:88:9F:25 57-9F-61-88-9F-25
[DEL] Device 5E:F5:F9:7C:6E:D7 5E-F5-F9-7C-6E-D7
```
</details>

I'd appreciate if you could clarify:
1. If I make this work at all (be it on another Linux distro, kernel version, BT dongle or Windows), would that mean that the Omron device now would be paired just to that new device, i.e. I won't be able to pull readings via mobile app?
2. Does it make sense to try and test it on Windows instead of Linux?
3. I'm total n00b when it comes to the Bluetooth on Linux, but are there other programming languages that rely on different underlying libraries that could make it work on the same OS version/BT adapter? As I understand, `bleak` relies on BlueZ, but IDK if that's the OS-level thingie or if there are other implementations.
4. Similar question, but regarding the MCUs. I wonder if that can work more reliably on ESP32, for instance. Sorry if that's a dumb question, but I never dug deep into BLE, and always assumed that it "just works", and the only issue with Bluetooth on Linux are missing drivers for some of the chipsets.

userx14, 2024-07-15:
> Here's how it looks like if I launch bluetoothctl in another terminal and respond yes to pairing.

Still seems to be the exact same problem.

> would that mean that the Omron device now would be paired just to that new device

You can just write re-pair it with the mobile app. When you use omblepy in the default mode, it will not modify the "currently unread recods" counter.

> Does it make sense to try and test it on Windows instead of Linux?

The Windows 10 bluetooth stack seems to be more reliable for BLE devices like the omron, so yes.

> I'm total n00b when it comes to the Bluetooth on Linux, but are there other programming languages that rely on different underlying libraries that could make it work on the same OS version/BT adapter?

So your main goal is to get it working on linux? 
I think the main problem is, that there is something fundamentally wrong with some versions of the linux bluetooth stack. 
No mater which tool/library on top you use (gatttool / bleak / bluezero), will not change anything. 
But I have not investigated what actually causes the problem of the missing services, 
maybe it is some mishap in the pairing process, or some bluez versions are just broken. 
The bluetooth pairing process is a bit complicated with all the possible variations:
(https://www.bluetooth.com/blog/bluetooth-pairing-part-1-pairing-feature-exchange/).

> Similar question, but regarding the MCUs. I wonder if that can work more reliably on ESP32, for instance. Sorry if that's a dumb question, but I never dug deep into BLE, and always assumed that it "just works", and the only issue with Bluetooth on Linux are missing drivers for some of the chipsets.

Yes the ESP32 would work, I already did some tests in this direction.
But so far, I did not develop if further. 
I could not find a nice way to transfer the data to the pc and give commands to the esp32. 
Writing the data to a SD card and then having to constantly swapp it between pc and esp seems troublesome. 
Maybe communication and dumping the data over serial interface would be an option. 
The esp32 could accept serial input to which mac address and using which device specifics it could connect. 
But developing code on the esp32 is harder, and debugging is not as nice as on pc. 

Maybe one could do an approach where the task is split between a pc application and the esp32 application:
The esp32 would implement the omron bluetooth communication, but which addresses are read from are specified by the pc.
Should be relativly simple to do the serial communication with pySerial.

For me it worked fine on windows 10, my tests for linux were only 10% successful. But win11 seems to break bluetooth compatibility, so I might have to explore the esp32 route sooner or later.

userx14, 2024-07-15:
This was very crude test from 2023, just for verifying if basic pairing could be done: [ombleESP32.ino.txt](https://github.com/user-attachments/files/16241270/ombleESP32.ino.txt)

maxromanovsky, 2024-07-16:
Thank you for the detailed answer.

> So your main goal is to get it working on linux?
What I want to achieve ideally, is sync readings to Garmin Connect via `omblepy`, but keep syncing the same readings to Omron Connect / Apple Health via native iOS app.

> I could not find a nice way to transfer the data to the pc and give commands to the esp32.
Common pattern is to use MQTT for 2-way communication.
Another option would be [ESPHome](https://esphome.io/#bluetooth-ble) with it's connectivity to Home Assistant.

RobertWojtowicz, 2024-07-23:
Hi @maxromanovsky, @userx14 

As if a working firmware for ESP32 would appear, I can test and implement in my integration as here:
https://github.com/RobertWojtowicz/export2garmin/blob/master/manuals/Miscale_ESP32.md

BR,
Robert

userx14, 2024-07-25:
Find the work in progress esp32 firmware [here](https://github.com/userx14/omblepy/tree/esp32bridge/esp32bridge).
In the current state it only supports pairing but not data transmission and has to be operated from the arduino serial monitor.

userx14, 2024-07-27:
The first working prototype of the esp32-bridge-firmware is uploaded in the branch. 
If anyone has time for testing, please report your results. For my HEM-7322 I could successfully read the data.

Thanks,
Benjamin

RobertWojtowicz, 2024-08-02:
Hi, @userx14 

I understand the issues of connection over COM in terms of pairing in the terminal, but will you create a version that sends MQTT data over WiFi ?

In many cases after my experience ESP32 modules have a small internal antenna which results in a small range therefore the gateway must be close to the bluetooth device.

For this reason I use a USB bluetooth with an external antenna so I have coverage throughout my apartment. ESP32 module makes sense when it can be embedded in the device or is very close to it, and the server is far away to be in WiFi range.

BR,
Robert

userx14, 2024-08-04:
> but will you create a version that sends MQTT data over WiFi ?

The issue would be that the full communication would need to be ported to the esp32. 
I find it much simpler to do the device specific communication and data handling in python, but there is no fundamental reason why it is not doable in c++ on the esp. Currently I do not have enough time to port more functionality to the esp32.

> In many cases after my experience ESP32 modules have a small internal antenna which results in a small range therefore the gateway must be close to the bluetooth device.

Some of them have external antenna connectors, which can increase the range like ESP32-WROOM-32UE. 
I agree that the standard modules are quiet limited in range.

Best,
Benjamin

RobertWojtowicz, 2024-08-05:
Hi, @userx14 

sure I understand, I also have a problem with time ;)
If anything, take a look at my code, maybe it will inspire in handling WIFI and MQTT.

BR,
Robert