userx14/omblepy #29: How to resolve: FutureWarning: BLEDevice.metadata is deprecated and will be removed in a future version of Bleak

RobertWojtowicz, 2024-07-18:
Hi, @userx14 

I completed version 1.0 of my project (https://github.com/RobertWojtowicz/export2garmin), using your module as well. So it touched me if it is not possible to remove some errors/warnings, maybe it is worth to lean over it, here is a suggestion how to improve it: https://github.com/hbldh/bleak/discussions/1354#discussioncomment-9811499

BR,
Robert

userx14, 2024-07-18:
Hi Robert,

thanks for the suggestion and linked issue, the warning should be fixed in the latest version.

Best
Benjamin

closed 2024-07-18 by userx14

RobertWojtowicz, 2024-07-19:
Hi, @userx14 

The fix works fine, I wonder what else to do with the non-serious type messages dumped:
```
   File "/home/robert/export2garmin-master/omron/omblepy.py", line 379,
in <module>
     asyncio.run(main())
```
```
  File "/home/robert/export2garmin-master/omron/omblepy.py", line 376,
in main
     await bleClient.unpair()
```
```	 
   File "/home/robert/export2garmin-master/omron/omblepy.py", line 377,
in main
     await bleClient.disconnect()
```
They do not affect the operation of omblepy in any way, I simply redirect them to the > /dev/null 2>&1

BR,
Robert

userx14, 2024-07-22:
Hi Robert,

actually I'm not really sure where this comes from. Is this on linux? 
Maybe the omron disconnects and the bluetooth client is destroyed before unpair() or disconnect() can be called, but that's just a guess. You are right that this should not affect the functionality.

Regards
Benjamin

RobertWojtowicz, 2024-07-22:
Hi @userx14

Yes, Linux Debian 12, but it looks like cosmetics, works without problems except for the warnings.

BR,
Robert