userx14/omblepy #30: I confirm the functioning of the Omron M4 Inteli IT :)

RobertWojtowicz, 2024-07-22:
@userx14 

Tested M4 Intelli IT works as M400 (HEM-7155T), can be added to matrix :)

BR,
Robert

userx14, 2024-07-22:
Hi Robert,

it's already there, but I really should order the rows so it's easier to find.
But it would be interesting for me to know with which platform you tested?

Regards
Benjamin

RobertWojtowicz, 2024-07-22:
Hi @userx14

In the matrix there is no information about the M4, I already take out, today successful tests were made on: 
Hardware: Raspberry PI 5 8GB, internal BT;
OS: Raspberry Pi OS (based on Debian 12, 64bit).

BR,
Robert

RobertWojtowicz, 2024-08-16:
@,

I confirm correct operation also with M7.

BR,
Robert

RobertWojtowicz, 2024-08-17:
@userx14 

I confirm correct operation also with M7 on this setup

BR,
Robert

userx14, 2024-08-17:
Thanks for testing,
I have added Raspberry Pi 4B and 5 with "Raspberry Pi OS (64-bit)" to the list of supported os / hardware.

RobertWojtowicz, 2024-08-18:
> Thanks for testing,
> I have added Raspberry Pi 4B and 5 with "Raspberry Pi OS (64-bit)" to the list of supported os / hardware.

@userx14 ,

You're welcome, I'm glad to be a part of this project :)

BR,
Robert

closed 2024-08-18 by RobertWojtowicz

RobertWojtowicz, 2024-09-10:
@userx14

To Omron device support matrix, in case of M7 Intelli IT, in section of testers add RobertWojtowicz.

BR,
Robert