userx14/omblepy #31: Pairing issue on HEM-7361T

Dream4Theater, 2024-07-29:
When pairing both windows 10 and ubuntu i have problems.

On windows 10;
Traceback (most recent call last):                                                                                                                                                                                                             File "C:\Users\sefa.kefeli\Desktop\bluetooth\omblepy\omblepy.py", line 390, in <module>                                                                                                                                                        asyncio.run(main())                                                                                                                                                                                                                        File "C:\Python310\lib\asyncio\runners.py", line 44, in run                                                                                                                                                                                    return loop.run_until_complete(main)                                                                                                                                                                                                       File "C:\Python310\lib\asyncio\base_events.py", line 646, in run_until_complete                                                                                                                                                                return future.result()                                                                                                                                                                                                                     File "C:\Users\sefa.kefeli\Desktop\bluetooth\omblepy\omblepy.py", line 365, in main                                                                                                                                                            await bleClient.pair(protection_level = 2)                                                                                                                                                                                                 File "C:\Python310\lib\site-packages\bleak\__init__.py", line 639, in pair                                                                                                                                                                     return await self._backend.pair(*args, **kwargs)                                                                                                                                                                                           File "C:\Python310\lib\site-packages\bleak\backends\winrt\client.py", line 623, in pair                                                                                                                                                        raise BleakError(f"Could not pair with device: {pairing_result.status}") 

On ubuntu;

sudo python3 ./omblepy.py -p -d HEM-7361T -m 00:5F:BF:7C:7A:2A
2024-07-29 15:15:06,746 - omblepy - INFO - Attempt to import module for device hem-7361t
2024-07-29 15:15:07,118 - omblepy - INFO - Attempt connecting to 00:5F:BF:7C:7A:2A.
2024-07-29 15:15:11,264 - omblepy - INFO - unpair and disconnect
Traceback (most recent call last):
  File "./omblepy.py", line 374, in main
    await bluetoothTxRxObj.writeNewUnlockKey()
  File "./omblepy.py", line 222, in writeNewUnlockKey
    raise ValueError(f"Could not enter key programming mode. Has the device been started in pairing mode? Got response: {deviceResponse}")
ValueError: Could not enter key programming mode. Has the device been started in pairing mode? Got response: bytearray(b'\x82\x0f\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00')

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "./omblepy.py", line 390, in <module>
    asyncio.run(main())
  File "/usr/lib/python3.8/asyncio/runners.py", line 44, in run
    return loop.run_until_complete(main)
  File "/usr/lib/python3.8/asyncio/base_events.py", line 616, in run_until_complete
    return future.result()
  File "./omblepy.py", line 388, in main
    await bleClient.disconnect()
  File "/usr/local/lib/python3.8/dist-packages/bleak/__init__.py", line 624, in disconnect
    return await self._backend.disconnect()
  File "/usr/local/lib/python3.8/dist-packages/bleak/backends/bluezdbus/client.py", line 425, in disconnect
    assert self._bus is None
AssertionError

ubuntu hostnameclt output is;

Icon name: computer
Machine ID: ***************
Boot ID: *************
Operating System: Ubuntu 20.04.5 LTS
Kernel: Linux 4.14.111
Architecture: arm

userx14, 2024-07-29:
Hi @Dream4Theater ,
did you try the (https://github.com/userx14/omblepy?tab=readme-ov-file#troubleshooting)[troubleshooting tips] from the readme? 
For the windows problem
- what is the bluetooth stick / adapter you are using (is bluetooth version new enough, is it a non harmony stack based dongle)
- do you get a pairing dialog from windows
- did you try to remove the paired bluetooth devices and test again? 
The error seems to come from non working bluetooth pairing from the os level, this is not device specific communication yet.

For the linux problem, it looks like the os bluetooth pairing is working, but the omron reponds with an error code when trying to write the unlock key. Reasons for this could be:
- previous unsucessful attempts (sometimes the omron device needs to be restarted so it works again, maybe remove the batteries)
- not in pairing mode (blinking P)
- phone with installed omron app also trying to communicate
But I have not tested with ubuntu 20.04, so there could also be other causes.

Regards,
Benjamin

Dream4Theater, 2024-07-30:
Thank you for your response,

So i needed to work that on linux so i didnt try to much on windows and on linux side i could manage to established to connection and now its working great. I dont know what cause the problem maybe too much unsucessful events caused it or i paired with my phone first than that caused the problem.

closed 2024-07-30 by Dream4Theater

userx14, 2024-07-30:
Glad you got it working 👍