userx14/omblepy #34: Adding hci selection

RobertWojtowicz, 2024-11-10:
Hi,
@userx14 

Is there an option to add a hci selection?

BR,
Robert

userx14, 2024-11-11:
Hi Robert,

could you go into more detail, I'm not sure if I got what you envision.
Do you want an option to switch between different Bluetooth Backends or between Bluetooth adapters?
For both of these things I will have to look into my currently used bluetooth libraray, and see if there is a way access this.

Regards
Benjamin

RobertWojtowicz, 2024-11-11:
Hi,
@userx14 

I mean bluetooth adapter, something like that ?
```parser.add_argument('-i', '--hci', type=str, help="Choose which HCI device you want to scan with.")```

```hci_device = args.hci if args.hci else None
bleClient = bleak.BleakClient(bleAddr, adapter=hci_device)```  ?

BR,
Robert

RobertWojtowicz, 2024-11-12:
Hi,
@userx14 

I tried but I don't know if it works, it just scans, I also checked the version below but it doesn't return me an error as it tries to cheat with the hci number:
```
parser.add_argument("-i", dest="adapter", default=None, help="HCI device")
```
```
bleClient = bleak.BleakClient(bleAddr, adapter=args.adapter)
```

Could you take a look at how it should look like ?

BR,
Robert

userx14, 2024-11-13:
Hi Robert,

maybe you can test if it works by calling the [`_get_adapter_path()`](https://github.com/hbldh/bleak/blob/e01e2640994b99066552b6161f84799712c396fa/bleak/backends/bluezdbus/client.py#L597-L611) function on your new client.
The adapter assignment seems to only work for the linux or android backend, I did not see any equivelent for windows.

Regards,
Benjamin

RobertWojtowicz, 2024-11-18:
Hi @userx14,

I added the HCI BLE adapter number selection in omblepy.py:
https://github.com/RobertWojtowicz/export2garmin/blob/master/omron/omblepy.py

BR,
Robert

closed 2024-11-18 by RobertWojtowicz

userx14, 2024-11-22:
Hi Robert,

thank you for testing out how to work with the adapter configuration. 
When I have the time I will try to include this in this repository too. 
For the command line argument parser I feel it is better to include it as a optional flag (so leaving out `required="true"`), 
since it is quiet a specialized use case with two bluetooth adapters and only applies to linux systems.

Best,
Benjamin

RobertWojtowicz, 2024-11-22:
Hi, @userx14 

enjoy, glad you liked concept :)

In my case it is required because I made the bluetooth diagnostics, and also needs to permanently declare hci or mac address of adapter (code will be ready in days, it is already working but I am still warming it up).

BR,
Robert

RobertWojtowicz, 2024-11-23:
@userx14 
I have released a new version: https://github.com/RobertWojtowicz/export2garmin/releases

BR,
Robert

userx14, 2024-11-23:
Hi Robert,

there is now a testing branch with the changes: 
https://github.com/userx14/omblepy/tree/hci-adapterTesting

Regards,
Benjamin