userx14/omblepy #35: HBF-222T-EBK Omron VIVA Body Composition

magcode, 2024-12-09:
Hello,

I bought the mentioned scale and I'm trying to add support for it.
So far I was able to pair the device using your code. Also it pulls data from it (using one of the blood pressure drivers) and writes it to the CSV file. So overall communication works.

Now the question is how to implement the "deviceSpecific_ParseRecordFormat". Would you have any hints how to tackle this?

I looked around in Github but it seems nobody has ever done this.
I tried to look into the Omron App but the decompiled code is very hard to understand.

How did you originally found all the details like "settingsReadAddress" or the structure of "singleRecordAsByteArray"?

Thank you!

userx14, 2024-12-09:
Hi magcode,

> So overall communication works.

It is possible that the devices reuses the generall bluetooth communication standard from omron, but the correct data could be stored at other eeprom addresses. 
Android has a feature to log all bluetooth transmissions in a log called btsnoop_hci.log which can generally give insights into the bluetooth transmissions between some app and device. But it is a bit cumbersome to extract it on some phones with adb and might require to generate a bugreport.zip on non rooted phones.

> I tried to look into the Omron App but the decompiled code is very hard to understand.

There is `omron_v7.5.apk/res/raw/batchdl_eu_001_006_00000.zip/Device/HBF`. 
But don't blindly trust the files ending in `*Index.json` or `*Indexes.json`, 
there seem to be some additional offsets missing.

> How did you originally found all the details like "settingsReadAddress"

For most devices there is a section called "setting index" inside DeviceConfig.sys

> or the structure of "singleRecordAsByteArray"?

If you already know how much bytes a record has, you can guess what the bits do, by comparing to known data values on the device. I guess they also did not reinvent the wheel for every appliance, so something like the date and time encoding will likely be simmilar.

Regards,
Benjamin

iddq, 2024-12-09:
Is this a blood pressure meter? Basically, the code expects the presence of a certain UUID, which primarily exists in BPMs and is referred to as the parent UUID in the code. Does this exist on your device? Do the pairing and unlock commands work as expected? If you don't have experience with this code, I would note initially that this is not just about Bluetooth pairing, but rather a kind of proprietary method as far as I understand, see the -p switch. This is about the program setting an unlocking key on the device.

userx14, 2024-12-09:
Hi iddq,
From what I gathered from the omron website, 
it is a digital scale with additional electrodes for measuring body composition with 50 kHz impedance analysis.
The communication protocol could be simmilar as it can store and transfer 30 records with data such as weight, muscle-fraction, etc.

magcode, 2024-12-11:
Thank you so much for the hints and pointers! I have enough to read and try now.

magcode, 2024-12-30:
Quick update: unfortunately not much progress.

I found the right parameters to read user1 ... user4 data in DeviceConfig.sys (Thank you!) and after adding a measurement on the device I do see a new value when pulling the data.

I'm stuck with interpreting the data. Here are my measurements as they enter "deviceSpecific_ParseRecordFormat"

```
62893a48699658183f97346c0ab81185072d00000000900000016280ffff4fb0
62893b08685655d83fab376c0a97123107401ec8072c900000026280ffff5ca3
62e937486de65f983fdc2b6cf42d0f6706d4000000009000000362e0ffff0af5
```

I cannot even find simple things like the year (should be 2024 in all cases)

userx14, 2024-12-31:
Hi magcode,

there could be some info in the VitalDataIndexes.json in the apk. But sometimes these offsets are not 100% correct.
> I cannot even find simple things like the year (should be 2024 in all cases)

The year is usually stored as an offset from 2000, so 2024=24=0x18. But it is possible, that the value only coveres the lower 6 bit, meaning that the byte here could contain the year:
```
62893a48699658 18 3f97346c0ab81185072d00000000900000016280ffff4fb0
62893b08685655 d8 3fab376c0a97123107401ec8072c900000026280ffff5ca3
62e937486de65f 98 3fdc2b6cf42d0f6706d4000000009000000362e0ffff0af5
```
For me it was possible to change the date / year locally on the omron device, and create another recording. This way it is much easier to spot the changes. Also it helps to know what measurement the data represents and compare against that. Otherwise it will be really difficult to reverse engineer the data location, from the record bytes alone.

Best,
Benjamin

magcode, 2025-01-01:
Thank you Benjamin,

I found a way to retrieve the data.
I'm using a service called "Record Access Control Point" and the code available [here](https://github.com/huraypositive/omron-android-sdk/blob/master/omronsdk/src/main/java/net/huray/omronsdk/ble/entity/internal/OmronMeasurementWS.java) to read/parse the records.

The process is quite different to omblepy structure thus I'll create my own little tool.

closed 2025-01-01 by magcode