userx14/omblepy #36: Omron HEM-7156T - update

mbalaatgit, 2024-12-29:
Hi,

Greetings.
I am new to this omron bluetooth integration.

I have the following update.

1. I tried with windows 11. To start with it was not possible to connect to windows bluetooth, probably due to timeout issue. ( save power check box ). Once configuration was done ( removed the check box). it started working.
2. I copied 7155T to 7156T. ( device specific)
3. I am now able to establish communication and the data reading happens and gets completed. Pair and unpair works consistently.

But have issues in saving the data, or creating the user.csv file. I am getting the error 2.
Even if I create user1.csv, the appendCSv function fails with backup... file not found.
Could you kindly help to resolve this?

Captured Log:
C:\Users\bala\Desktop\omblepy-main\omblepy-main>python omblepy.py -n -d HEM-7156T --loggerDebug
2024-12-29 19:32:02,287 - omblepy - INFO - Attempt to import module for device hem-7156t
To improve your chance of a successful connection please do the following:
 -remove previous device pairings in your OS's bluetooth dialog
 -enable bluetooth on you omron device and use the specified mode (pairing or normal)
 -do not accept any pairing dialog until you selected your device in the following list

Select your Omron device from the list below...
+----+-------------------+-------------------------------+------+
| ID | MAC               | NAME                          | RSSI |
+----+-------------------+-------------------------------+------+
| 0  | E3:B7:B1:D4:FC:EA | None                          | -50  |
| 1  | 00:5F:BF:45:D8:B9 | BLEsmart_0000057A005FBF45D8B9 | -54  |
| 2  | 49:EC:69:7A:93:C6 | None                          | -93  |
| 3  | D9:7A:DA:B6:1F:10 | None                          | -99  |
+----+-------------------+-------------------------------+------+
Enter ID or just press Enter to rescan.
1
2024-12-29 19:32:11,131 - omblepy - INFO - Attempt connecting to 00:5F:BF:45:D8:B9.
2024-12-29 19:32:15,567 - omblepy - INFO - communication started
2024-12-29 19:32:17,366 - omblepy - DEBUG - tx ch0 > 0800000000100018
2024-12-29 19:32:17,461 - omblepy - DEBUG - rx ch0 < 18800000001000000094057affffffff
2024-12-29 19:32:17,461 - omblepy - DEBUG - rx ch1 < ffffffff0bff00970000000000000000
2024-12-29 19:32:17,505 - omblepy - DEBUG - read from 0x10 size 0x10
2024-12-29 19:32:17,505 - omblepy - DEBUG - tx ch0 > 0801000010100009
2024-12-29 19:32:17,598 - omblepy - DEBUG - rx ch0 < 1881000010100d000080008000800d00
2024-12-29 19:32:17,612 - omblepy - DEBUG - rx ch1 < 00000000008000990000000000000000
2024-12-29 19:32:17,644 - omblepy - DEBUG - read from 0x3c size 0x10
2024-12-29 19:32:17,644 - omblepy - DEBUG - tx ch0 > 080100003c100025
2024-12-29 19:32:17,749 - omblepy - DEBUG - rx ch0 < 188100003c10c8a8000000000000180c
2024-12-29 19:32:17,750 - omblepy - DEBUG - rx ch1 < 1d13170fea00003d0000000000000000
2024-12-29 19:32:17,786 - omblepy - INFO - Current ring buffer slot user1: 13.
2024-12-29 19:32:17,786 - omblepy - INFO - Unread records user1: 0.
2024-12-29 19:32:17,786 - omblepy - INFO - Current ring buffer slot user2: 0.
2024-12-29 19:32:17,786 - omblepy - INFO - Unread records user2: 0.
2024-12-29 19:32:17,787 - omblepy - INFO - start reading data, this can take a while, use debug flag to see progress
2024-12-29 19:32:17,787 - omblepy - DEBUG - write to 0x54 size 0x10
2024-12-29 19:32:17,787 - omblepy - DEBUG - tx ch0 > 1801c00054100d000080008000800d00
2024-12-29 19:32:17,806 - omblepy - DEBUG - tx ch1 > 000000000080009d
2024-12-29 19:32:17,901 - omblepy - DEBUG - rx ch0 < 1881c00054100d000080008000800d00
2024-12-29 19:32:17,902 - omblepy - DEBUG - rx ch1 < 000000000080001d0000000000000000
2024-12-29 19:32:17,939 - omblepy - DEBUG - tx ch0 > 080f000000000007
2024-12-29 19:32:18,025 - omblepy - DEBUG - rx ch0 < 088f0000000000870000000000000000
2024-12-29 19:32:18,178 - omblepy - INFO - communication finished
2024-12-29 19:32:18,179 - omblepy - INFO - unpair and disconnect
Traceback (most recent call last):
  File "C:\Users\bala\Desktop\omblepy-main\omblepy-main\omblepy.py", line 395, in <module>
    asyncio.run(main())
  File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.12_3.12.2288.0_x64__qbz5n2kfra8p0\Lib\asyncio\runners.py", line 194, in run
    return runner.run(main)
           ^^^^^^^^^^^^^^^^
  File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.12_3.12.2288.0_x64__qbz5n2kfra8p0\Lib\asyncio\runners.py", line 118, in run
    return self._loop.run_until_complete(task)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.12_3.12.2288.0_x64__qbz5n2kfra8p0\Lib\asyncio\base_events.py", line 686, in run_until_complete
    return future.result()
           ^^^^^^^^^^^^^^^
  File "C:\Users\bala\Desktop\omblepy-main\omblepy-main\omblepy.py", line 382, in main
    appendCsv(allRecs)
  File "C:\Users\bala\Desktop\omblepy-main\omblepy-main\omblepy.py", line 267, in appendCsv
    backup.write_bytes(oldCsvFile.read_bytes())
  File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.12_3.12.2288.0_x64__qbz5n2kfra8p0\Lib\pathlib.py", line 1036, in write_bytes
    with self.open(mode='wb') as f:
         ^^^^^^^^^^^^^^^^^^^^
  File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.12_3.12.2288.0_x64__qbz5n2kfra8p0\Lib\pathlib.py", line 1013, in open
    return io.open(self, mode, buffering, encoding, errors, newline)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
FileNotFoundError: [Errno 2] No such file or directory: 'backup_user1_2024_12_29__19_32_18.csv'

Regards
Bala.

userx14, 2024-12-31:
Hi bala,

my first observation is that you run the tool with the -n flag. This will only read new records and also resets the counter for the unread records to zero. In the log I see that there is some data records for user1, but they are not read back, because they are not new records: "Unread records user1: 0.". So I suggest to try it again without this flag.

As for the error you see for creating the backup file, I'm not really sure where it comes from. What the tool tries to do there is to create a byte-for-byte identical copy of the old csv file, and write it to a newly created file called "backup_userX_currentDate.csv". It should automatically create this file when calling write_bytes in your current working directory and write to it. Could this be a permission problem with writing into your working directory? 
As a workaround you could comment and disable the line that creates the backup file here: https://github.com/userx14/omblepy/blob/752de92f621aade76609b0316633bfb52847d3bd/omblepy.py#L267

Best
Benjamin

mbalaatgit, 2025-01-01:
Hi Benjamin,

Thanks for your input.

The issue has been resolved. 

The error is due to Windows 11 ransomware protection. Allowing Python.exe in the settings, resolves this issue. There is no No code issue.

All working well.

Appreciate your effort in creating this library.

regards
Bala.