userx14/omblepy #38: OS Suuport for MacOS mis-interpreted by Grok

chengguan, 2025-01-22:
Grok told me that this project supports MacOS and I believe the issue could have been the OS support column uses emoji instead of "Yes/No/Partial".

I would like to suggest that human readable form to be added to the OS support column so that AI models can interpret it more accurately.

Thank you very much.

Regards,
CG.

Xtrem532, 2025-01-22:
Live by the AI Brainrot, Die by the AI Brainrot

m4m1de, 2025-01-22:
Please do not replace your brain or your thinking or information gathering process with a Language model.

userx14, 2025-01-22:
Actually there might be a way to get it working with mac. 
The [esp32 version](https://github.com/userx14/omblepy/tree/esp32bridge) on the bottom of the table works independent of the pc bluetooth stack, but requires additional hardware. I have no tested it on mac os, my main concern would be if there is a supported usb serial driver for these boards.

userx14, 2025-01-22:
For the table entry, I actually prefer to keep color coded symbols. 
Imho it enables a quick lookup for some of the more complicated usage options in the tables, here for comparison without icons:
[image]
I would also like to keep the style somewhat consistent for all the tables. 
In the end it is a tradeoff between fast visual understanding vs. machine readability, where I would lean towards the human usage.

closed 2025-01-22 by userx14