userx14/omblepy #39: Could you add support to the HEM-7140T1?

Fadel-K, 2025-03-15:
I would love it if you could add support to HEM-7140T1.

I have attached the btsnoop file from my phone while it ran the omron connect app.

[btsnoop_hci.log](https://github.com/user-attachments/files/19260282/btsnoop_hci.log)

userx14, 2025-03-28:
Hi  @Fadel-K ,

sorry it took me quiet some time to get to this. The communication in your log looks like the one from the other omron devices I worked with. Here is my initial decoding attempt:
```
BT-Communication, response indented
08 0000 0000 10 0018                                            #start communication
    08 8000 0000 10 0098
08 0100 0260 2c 0047                                            #settings read
    34 8100 0260 2c 00400080008000802100008000000080000101000a000000000000000c00ffff00000000000000000000fe00 00e2
08 0100 028c 18 009f                                            #settings read
    20 8100 028c 18 c8a800000000000019030e0e0b10c3000000000000000000 0095
08 0100 02e8 38 00db                                            #likely data
    40 8100 02e8 38 584e4419ac0d1e1600001400040058494119ac0d651700001500450054494119ac0df21700001600cf0076756219ac0d4c18000017009a00 00f3
08 0100 0320 38 0012                                            #likely data
    40 8100 0320 38 5b4c4619ac0da9180000180098005f504419ac0df81c00001900f2005c474119ac0dda1d00001a00c7005e4b4719ac0d991e00001b009400 00e4
08 0100 0358 38 006a                                            #likely data
    40 8100 0358 38 5a4f4519ad0d421000001c002f005c484419ad0d821100001d006b005d503b19ad0d6d1200001e00580064513c19ad0d0f1500001f000700 0022
08 0100 0390 1c 0086                                            #likely data
    24 8100 0390 1c 58504519ae0d681c0000200065005a513e19cb0d4a1b000021006000 0016
18 01c0 02a4 10 00400080008000802100008000000080 008e           #does not change values
    18 81c0 02a4 10 00400080008000802100008000000080 000e
18 01c0 02d0 10 c8a800000000000019030e0e0b11c400 00bf           #only changes 10c300->11c400, maybe time sync update?
    18 81c0 02d0 10 c8a800000000000019030e0e0b11c400 003f       
08 0f00 0000 00 0007                                            #end communication
    088f000000000087
```

If you have the time please test the branch I made for testing: https://github.com/userx14/omblepy/compare/hem-7140t-testing and check if it works.

Best regards,
Benjamin

Fadel-K, 2025-03-30:
Thank you for your reply @userx14 ! I am unable to pair the monitor using preexisting models on my RPi running Ubuntu 22.04.5 LTS. it gave the following error on the terminal:
```

  File "/home/fadel/Downloads/omblepy-main/./omblepy.py", line 395, in <module>
    asyncio.run(main())
  File "/usr/lib/python3.10/asyncio/runners.py", line 44, in run
    return loop.run_until_complete(main)
  File "/usr/lib/python3.10/asyncio/base_events.py", line 649, in run_until_complete
    return future.result()
  File "/home/fadel/Downloads/omblepy-main/./omblepy.py", line 364, in main
    await bleClient.pair(protection_level = 2)
  File "/home/fadel/.local/lib/python3.10/site-packages/bleak/__init__.py", line 629, in pair
    return await self._backend.pair(*args, **kwargs)
  File "/home/fadel/.local/lib/python3.10/site-packages/bleak/backends/bluezdbus/client.py", line 473, in pair
    assert_reply(reply)
  File "/home/fadel/.local/lib/python3.10/site-packages/bleak/backends/bluezdbus/utils.py", line 22, in assert_reply
    raise BleakDBusError(reply.error_name, reply.body)
bleak.exc.BleakDBusError: [org.bluez.Error.AuthenticationCanceled] Authentication Canceled

```

Can you guide me on further steps? Thanks for your help!

userx14, 2025-04-01:
The error you are seeing happens during the pairing of the bluetooth devices and comes from the bluez backend.
It can be caused by two things:
- For the initial pairing for the devices I've tested with, one has to confirm a pairing dialog on the os. However some linux distribution do not correctly display it. There is some info in the linux part of the [troubleshooting section](https://github.com/userx14/omblepy?tab=readme-ov-file#troubleshooting) about using bluetoothctl in a second terminal to get it working. Do you get a pairing request dialog on your pi?
- With some linux versions (my suspicion is that it is connected to the bluez version) I just was not able to get the pairing process working. I think it was also ubuntu 22 and I did unfortunately not find a fix for this, other than to use a different operating system / version. Some people reported that it worked with their pi 4b using raspberry pi os.

One alternative could also be the [esp32 version](https://github.com/userx14/omblepy/tree/esp32bridge?tab=readme-ov-file), where the OS specific bluetooth stack is bypassed by using a esp32 board to do the bluetooth pairing and communication.

If you have a win10 device I would recommend to do the first tests with it. 
The pairing dialog has a very short timeout and needs to be accepted quickly, but in general it works more reliably.

Best regards,
Benjamin

Fadel-K, 2025-04-01:
Thank you for your response. I tried the troubleshooting section about using bluetoothhctl in a 2nd terminal, I didn't get any pairing request on it. 

I am reluctant to change to raspbian os. Do you know which versions of Ubuntu/bluez are known to work? I don't have a win 10 device sadly, but I will try to get acccess to one and update if I am successful.

userx14, 2025-04-07:
I did not test so much with Ubuntu, the only thing I remember is that 20.04 was not able to pair with the device. On an older debian image, I had success with bluez 5.55.