userx14/omblepy #41: Could you add support for HEM-7361-ESL?

user0815872, 2025-04-07:
Message from omblepy:
D:\Tools\omblepy-main>python ./omblepy.py -p -d HEM-7361T
2025-04-07 12:39:47,402 - omblepy - INFO - Attempt to import module for device hem-7361t
To improve your chance of a successful connection please do the following:
 -remove previous device pairings in your OS's bluetooth dialog
 -enable bluetooth on you omron device and use the specified mode (pairing or normal)
 -do not accept any pairing dialog until you selected your device in the following list

Select your Omron device from the list below...
+----+-----+------+------+
| ID | MAC | NAME | RSSI |
+----+-----+------+------+
Enter ID or just press Enter to rescan.

user0815872, 2025-04-07:
System has recognized the device. I got the Same Message with paired and with unpaired device. I will try with another Bluetooth dongle.

userx14, 2025-06-11:
Please reopen, in case it does not work.