userx14/omblepy #44: [feature request] PlatformIO compatibility

gotnone, 2025-06-11:
Fixes to permit ease of importing into a PlatformIO project.

PlatformIO needs to have:

1. `#include <Ardino.h>` at the top
2. Needs correct ordering of function definitions or functions declared prior to use

userx14, 2025-07-06:
Should now be compatible with both platformIO and arduinoIDE which commit https://github.com/userx14/omblepy/commit/e71015fa888e8109a8de6e0e53a9974e3c3d6bff. 
It was a bit difficult to use the same code, since by default they use different versions of the arduino-esp32 core, which has breaking changes like the return type:

In the older version case `bleScanP->start` returns by value, while in the new one it returns a reference.
This is likely the reason why it did not compile here https://github.com/userx14/omblepy/commit/513ce1259649b42b898bf39af003ca9c50aae74d. 
Changed it back to a pointer and set the arduino-esp32 core to 3.x to have the same behaviour for both arduinoIDE and platformIO.

gotnone, 2025-07-07:
If you wan to explicitly use a return type, it is also possible to to do something like the following:

```
loop() {
...
    case 's':{
...
      BLEScan* bleScanP = BLEDevice::getScan();
      Serial.printf("s [");
      using start_t = decltype(bleScanP->start(1, false));
      [[maybe_unused]] start_t foundDevicesP = bleScanP->start(1, false);
...
```

I think `auto` is more C++11 idiomatic.  It also avoids any possibility of getting out of sync between the using alias and the function call parameters / overload function return type.

Because the result isn't currently used does it really matter what type is returned?

userx14, 2025-07-07:
> What do you think of making the return type `auto`? Do you think that would work with both arduino-esp32 core versions that seem to be floating around?

Good suggestion, completely forgot that `auto` exists, added in https://github.com/userx14/omblepy/commit/afb35ff0ed603348b0b8b2437b0881569d555899, thank you.