userx14/omblepy #54: X4 smart

Nwanko, 2025-08-02:
Any chance that the X4 Smart gets added to the support list?

userx14, 2025-08-02:
X4 Smart is corresponding to hem-7155t-esl. So there is a very high chance it already works with the hem-7155t driver.

Nwanko, 2025-08-02:
I get this back:
4
2025-08-02 15:30:03,167 - omblepy - INFO - Attempt connecting to 00:5F:BF:F3:B8:5E.
2025-08-02 15:30:16,632 - omblepy - INFO - unpair and disconnect
Traceback (most recent call last):
  File "/usr/local/lib/python3.11/dist-packages/bleak/backends/bluezdbus/client.py", line 297, in connect
    reply = await self._bus.call(
            ^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/dbus_fast/aio/message_bus.py", line 398, in call
    await future
asyncio.exceptions.CancelledError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/Nvvanko/export2garmin-master/omron/omblepy.py", line 403, in <module>
    asyncio.run(main())
  File "/usr/lib/python3.11/asyncio/runners.py", line 190, in run
    return runner.run(main)
           ^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/asyncio/runners.py", line 118, in run
    return self._loop.run_until_complete(task)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/asyncio/base_events.py", line 653, in run_until_complete
    return future.result()
           ^^^^^^^^^^^^^^^
  File "/home/Nvvanko/export2garmin-master/omron/omblepy.py", line 371, in main
    await bleClient.connect()
  File "/usr/local/lib/python3.11/dist-packages/bleak/__init__.py", line 566, in connect
    await self._backend.connect(self._pair_before_connect, **kwargs)
  File "/usr/local/lib/python3.11/dist-packages/bleak/backends/bluezdbus/client.py", line 154, in connect
    async with async_timeout(timeout):
  File "/usr/lib/python3.11/asyncio/timeouts.py", line 98, in __aexit__
    raise TimeoutError
TimeoutError

Nwanko, 2025-08-02:
Got this now : 
02.08.2025-16:20:45 SYSTEM * BLE adapter check if available
02.08.2025-16:20:52 SYSTEM * BLE adapter hci0(E4:5F:01:76:3B:35) working, go to pairing
2025-08-02 16:20:53,023 - omblepy - INFO - Attempt to import module for device hem-7155t
To improve your chance of a successful connection please do the following:
 -remove previous device pairings in your OS's bluetooth dialog
 -enable bluetooth on you omron device and use the specified mode (pairing or normal)
 -do not accept any pairing dialog until you selected your device in the following list

Select your Omron device from the list below...
+----+-------------------+-------------------------------+------+
| ID | MAC               | NAME                          | RSSI |
+----+-------------------+-------------------------------+------+
| 0  | 78:9C:50:01:03:5B | LEDBLE-01035B                 | -40  |
| 1  | CC:4D:75:C7:1D:AA | Xiaomi Scale S400 1DAA        | -41  |
| 2  | 00:5F:BF:F3:B8:5E | BLEsmart_00000265005FBFF3B85E | -43  |
| 3  | 68:AF:AC:2D:48:F3 | None                          | -43  |
| 4  | 0D:E2:EE:6C:B6:26 | None                          | -57  |
| 5  | 0E:6D:22:B4:78:84 | None                          | -58  |
| 6  | 42:A2:6E:B8:80:AA | OLED55CX3LA                   | -71  |
| 7  | 50:2C:C6:AF:51:DA | GR-AC_10001_09_51da_SC        | -75  |
| 8  | 4C:BC:E9:FA:FB:DF | None                          | -78  |
| 9  | 44:CB:8B:F6:71:4B | None                          | -83  |
| 10 | 4C:BC:E9:C4:D2:B1 | None                          | -102 |
+----+-------------------+-------------------------------+------+
Enter ID or just press Enter to rescan.
2
2025-08-02 16:21:01,680 - omblepy - INFO - Attempt connecting to 00:5F:BF:F3:B8:5E.
2025-08-02 16:21:10,551 - omblepy - INFO - Paired device successfully with new key bytearray(b'\xde\xad\xbe\xaf\x124\x124\xde\xad\xbe\xaf\x124\x124').
2025-08-02 16:21:10,552 - omblepy - INFO - From now on you can connect omit the -p flag, even on other PCs with different bluetooth-mac-addresses.
2025-08-02 16:21:10,835 - omblepy - DEBUG - tx ch0 > 0800000000100018
2025-08-02 16:21:10,955 - omblepy - DEBUG - rx ch0 < 188000000010000000940265ffffffff
2025-08-02 16:21:10,957 - omblepy - DEBUG - rx ch1 < fffffffff3ff00770000000000000000
2025-08-02 16:21:10,995 - omblepy - DEBUG - tx ch0 > 080f000000000007
2025-08-02 16:21:11,105 - omblepy - DEBUG - rx ch0 < 088f0000000000870000000000000000
2025-08-02 16:21:11,164 - omblepy - INFO - unpair and disconnect
2025-08-02 16:21:13,716 - omblepy - ERROR - Bleak AssertionError during disconnect. This usually happens when using the bluezdbus adapter.
2025-08-02 16:21:13,716 - omblepy - ERROR - You can find the upstream issue at: https://github.com/hbldh/bleak/issues/641
2025-08-02 16:21:13,717 - omblepy - ERROR - AssertionError details:

userx14, 2025-08-05:
Thank you for testing, added to support matrix 👍 .

Nwanko, 2025-08-06:
> Ok, [@userx14](https://github.com/userx14), we have confirmation that it works and can be added to the matrix :) [RobertWojtowicz/export2garmin#61](https://github.com/RobertWojtowicz/export2garmin/discussions/61)
> 
> [@Nwanko](https://github.com/Nwanko), I have released a new version of parallel scanning (experimental, I haven't tested yet because I don't have the time). Applies to the S400 and Omron use variant: https://github.com/RobertWojtowicz/export2garmin/blob/master/manuals/about_BLE.md#264-using-two-ble-adapters-in-parallel-experimental-version
> 
> BR, Robert

I don't have 2 bt adapters. I'm on rpi zero w 2 , i'll buy one later. does the pi have support for usb adapters?