userx14/omblepy #55: Support for HEM-7144T2

TreeviewGera, 2025-08-14:
Hey. I'd like to get support added for version 7144T2. I've been trying to use the script with it but even getting it to recognize a successful pairing hasn't been possible.

Here's the output from `service_explorer.py`:

```
2025-08-14 16:23:28,950 __main__ INFO: starting scan...
2025-08-14 16:23:29,331 __main__ INFO: connecting to device...
2025-08-14 16:23:29,586 __main__ INFO: connected to BLESmart_000000BCF95D33C7649F (F9:5D:33:C7:64:9F)
2025-08-14 16:23:29,586 __main__ INFO: [Service] 00001800-0000-1000-8000-00805f9b34fb (Handle: 1): Generic Access Profile
2025-08-14 16:23:29,943 __main__ INFO:   [Characteristic] 00002a00-0000-1000-8000-00805f9b34fb (Handle: 2): Device Name (read), Value: bytearray(b'HEM-7144T2')
2025-08-14 16:23:30,002 __main__ INFO:   [Characteristic] 00002a01-0000-1000-8000-00805f9b34fb (Handle: 4): Appearance (read), Value: bytearray(b'\x81\x03')
2025-08-14 16:23:30,092 __main__ INFO:   [Characteristic] 00002a04-0000-1000-8000-00805f9b34fb (Handle: 6): Peripheral Preferred Connection Parameters (read), Value: bytearray(b'\x10\x00\x18\x00\x00\x00\xc8\x00')
2025-08-14 16:23:30,093 __main__ INFO: [Service] 00001801-0000-1000-8000-00805f9b34fb (Handle: 8): Generic Attribute Profile
2025-08-14 16:23:30,152 __main__ INFO:   [Characteristic] 00002a05-0000-1000-8000-00805f9b34fb (Handle: 9): Service Changed (read,indicate), Value: bytearray(b'\x01\x00\xff\xff')
2025-08-14 16:23:30,212 __main__ INFO:     [Descriptor] 00002902-0000-1000-8000-00805f9b34fb (Handle: 11): Client Characteristic Configuration, Value: bytearray(b'\x02\x00')
2025-08-14 16:23:30,212 __main__ INFO: [Service] 0000180a-0000-1000-8000-00805f9b34fb (Handle: 12): Device Information
2025-08-14 16:23:30,272 __main__ INFO:   [Characteristic] 00002a29-0000-1000-8000-00805f9b34fb (Handle: 13): Manufacturer Name String (read), Value: bytearray(b'OMRONHEALTHCARE')
2025-08-14 16:23:30,332 __main__ INFO:   [Characteristic] 00002a24-0000-1000-8000-00805f9b34fb (Handle: 15): Model Number String (read), Value: bytearray(b'HEM-7144T2')
2025-08-14 16:23:30,392 __main__ INFO:   [Characteristic] 00002a25-0000-1000-8000-00805f9b34fb (Handle: 17): Serial Number String (read), Value: bytearray(b'0000021F0200C730D816')
2025-08-14 16:23:30,452 __main__ INFO:   [Characteristic] 00002a26-0000-1000-8000-00805f9b34fb (Handle: 19): Firmware Revision String (read), Value: bytearray(b'SI.S01.16.02')
2025-08-14 16:23:30,512 __main__ INFO:   [Characteristic] 00002a28-0000-1000-8000-00805f9b34fb (Handle: 21): Software Revision String (read), Value: bytearray(b'0000000000105001')
2025-08-14 16:23:30,573 __main__ INFO:   [Characteristic] 00002a23-0000-1000-8000-00805f9b34fb (Handle: 23): System ID (read), Value: bytearray(b'\xff\xff\xff\xff\xff\xff\xff\xff')
2025-08-14 16:23:30,573 __main__ INFO: [Service] 0000fe4a-0000-1000-8000-00805f9b34fb (Handle: 25): OMRON HEALTHCARE Co.: Ltd.
2025-08-14 16:23:30,632 __main__ INFO:   [Characteristic] b305b680-aee7-11e1-a730-0002a5d5c51b (Handle: 26): Unknown (read,write-without-response,notify,write), Value: bytearray(b'\x11\xacy\x9e\xa7\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'), Max write w/o rsp size: 64
2025-08-14 16:23:30,722 __main__ INFO:     [Descriptor] 00002902-0000-1000-8000-00805f9b34fb (Handle: 28): Client Characteristic Configuration, Value: bytearray(b'\x00\x00')
2025-08-14 16:23:30,722 __main__ INFO:   [Characteristic] db5b55e0-aee7-11e1-965e-0002a5d5c51b (Handle: 29): Unknown (write-without-response,write), Max write w/o rsp size: 64
2025-08-14 16:23:30,842 __main__ INFO:   [Characteristic] 49123040-aee8-11e1-a74d-0002a5d5c51b (Handle: 31): Unknown (read,notify), Value: bytearray(b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00')
2025-08-14 16:23:30,902 __main__ INFO:     [Descriptor] 00002902-0000-1000-8000-00805f9b34fb (Handle: 33): Client Characteristic Configuration, Value: bytearray(b'\x00\x00')
2025-08-14 16:23:30,902 __main__ INFO: [Service] 00001805-0000-1000-8000-00805f9b34fb (Handle: 34): Current Time Service
2025-08-14 16:23:30,962 __main__ INFO:   [Characteristic] 00002a2b-0000-1000-8000-00805f9b34fb (Handle: 35): Current Time (read,write,notify), Value: bytearray(b'\xd0\x07\x01\x01\x00\x01;\x00\x00\x01')
2025-08-14 16:23:31,022 __main__ INFO:     [Descriptor] 00002902-0000-1000-8000-00805f9b34fb (Handle: 37): Client Characteristic Configuration, Value: bytearray(b'\x00\x00')
2025-08-14 16:23:31,022 __main__ INFO: [Service] 00001810-0000-1000-8000-00805f9b34fb (Handle: 38): Blood Pressure
2025-08-14 16:23:31,022 __main__ INFO:   [Characteristic] 00002a35-0000-1000-8000-00805f9b34fb (Handle: 39): Blood Pressure Measurement (indicate)
2025-08-14 16:23:31,082 __main__ INFO:     [Descriptor] 00002902-0000-1000-8000-00805f9b34fb (Handle: 41): Client Characteristic Configuration, Value: bytearray(b'\x00\x00')
2025-08-14 16:23:31,142 __main__ INFO:   [Characteristic] 00002a49-0000-1000-8000-00805f9b34fb (Handle: 42): Blood Pressure Feature (read), Value: bytearray(b'\x07\x00')
2025-08-14 16:23:31,142 __main__ INFO: disconnecting...
2025-08-14 16:23:31,259 __main__ INFO: disconnected
```

And here's a btsnoop log, filtered to only include communication between my phone and the device:
[hem_7144t2.log](https://github.com/user-attachments/files/21775952/hem_7144t2.log)

Right off the bat you can see there doesn't seem to be a service matching with
```python
parentService_UUID        = "ecbe3980-c9a2-11e1-b1bd-0002a5d5c51b"
```
so I had to remove the check for that during my tests.

I'm using Windows 10 by the way. I'd appreciate any help you can provide on this.

TreeviewGera, 2025-08-15:
Hey Benjamin, thank you so much for your quick response.

I did as you suggested and wrote the file as such:

```python
import sys
import datetime
import logging
logger = logging.getLogger("omblepy")

sys.path.append('..')
from sharedDriver import sharedDeviceDriverCode

class deviceSpecificDriver(sharedDeviceDriverCode):

    deviceEndianess           = "big"
    userStartAdressesList     = [0x2e8]
    perUserRecordsCountList   = [30]
    recordByteSize            = 0x0e
    transmissionBlockSize     = 0x38

    settingsReadAddress       = 0x260
    #settingsWriteAddress            = 0x0286

    #AsettingsUnreadRecordsBytes      = [0x00, 0x08]
    #AsettingsTimeSyncBytes           = [0x14, 0x1e]

    def deviceSpecific_ParseRecordFormat(self, singleRecordAsByteArray):
        #recordDict["mov"]      = self._bytearrayBitsToInt(singleRecordAsByteArray, 32, 32)
        #recordDict["ihb"]      = self._bytearrayBitsToInt(singleRecordAsByteArray, 33, 33)

        recordDict             = dict()
        recordDict["sys"]      = self._bytearrayBitsToInt(singleRecordAsByteArray, 0, 7) + 25
        recordDict["dia"]      = self._bytearrayBitsToInt(singleRecordAsByteArray, 8, 15)
        recordDict["bpm"]      = self._bytearrayBitsToInt(singleRecordAsByteArray, 16, 23)
        year                   = self._bytearrayBitsToInt(singleRecordAsByteArray, 24, 31) + 2000
        month                  = self._bytearrayBitsToInt(singleRecordAsByteArray, 32, 35)
        day                    = self._bytearrayBitsToInt(singleRecordAsByteArray, 36, 40)
        hour                   = self._bytearrayBitsToInt(singleRecordAsByteArray, 41, 45)
        minute                 = self._bytearrayBitsToInt(singleRecordAsByteArray, 50, 55)
        second                 = self._bytearrayBitsToInt(singleRecordAsByteArray, 56, 61)
        second                 = min([second, 59])
        recordDict["datetime"] = datetime.datetime(year, month, day, hour, minute, second)
        return recordDict

    def deviceSpecific_syncWithSystemTime(self):
        raise ValueError("Not supported yet.")
```

This may be correct or not, the issue is I can't get the pairing to fully complete to test.
I'm running the following command:
```
python313 omblepy.py -d HEM-7144T2 -p -m F9:XX:XX:XX:XX:XX
```

But the script hangs indefinitely after printing `omblepy - INFO - Attempt connecting to F9:XX:XX:XX:XX:XX.` The device stays in pairing mode, with the blinking "P".

Adding some more `print` calls to check where exactly this happens, I find out that execution reaches this line:
```
        await bleClient.write_gatt_char(self.deviceUnlock_UUID, b'\x02' + b'\x00', response=True)
```
but the while loop right after never finishes. Indeed adding a print statement to the `_callbackForUnlockChannel` methods confirms that it never does get called. I'm not very familiar with bluetooth protocols so I'm not sure why this happens. Do you have any ideas on what could the issue be?

Thanks again.

userx14, 2025-08-15:
Do you get a pairing dialog in windows? 
In my past tests being quick when accepting the dialog was essential for it working.
If it freezes when writing to a characteristic I would guess it is a problem with bluetooth pairing not working.
For my tests I always delete the device out of the list of previously paired devices, have the bt-settings open so the pairing dialog comes up more quickly and accept within 1-2 seconds.

There would be an option to either test it with esp32 or an android phone to rule out windows or bluetooth hardware issues. 
For esp32: https://github.com/userx14/omblepy/tree/esp32bridge 
For android with nrfconnect: https://github.com/userx14/omblepy/issues/16#issuecomment-1657201052 .

TreeviewGera, 2025-08-15:
Yes, the pairing dialog does appear, and I do accept it really fast, within 1 or 2 seconds.

Sadly I do not have an esp32 to test with.

I tried nrfconnect. First I replaced all instances of `ecbe3980-c9a2-11e1-b1bd-0002a5d5c51b` with `0000fe4a-0000-1000-8000-00805f9b34fb` in the xml. This allowed me to run the macro. All `write` steps seemed to work fine and displayed a checkmark when run, but the `read` steps showed an alert icon.

userx14, 2025-08-16:
Does the blinking -P- disapear after you try with nrfconnect?
Maybe the pairing process is different for this device, would it be possible to record a hcilog of this?

TreeviewGera, 2025-08-16:
The blinking P does not disappear, no.
The log I shared above should start with the pairing process, via the official app. Is that what you mean?

userx14, 2025-08-17:
From the 0x11 start of the message to total message should be 17 bytes long, and the key should be 16 bytes long. 
Not sure why this does not work. The message in the hci log has a total length of 20 bytes which is also strange.

The "Write Request" you see in the log is the same as `bleClient.start_notify(self.deviceUnlock_UUID, ...)`. This will notify the client when there is a change in the characteristic and generate a callback, this happens in packet 115 "Value Notification". I also see this happening for the other omron devices, where the communication works. You can try removing the start_notify and just repeatedly read the characteristic instead as a test if this is the part of the code where it freezes the python script.

The nrfconnect app was very helpful in my testing, you can connect to the device without having to quickly confirm the pairing dialog, enable "notify on change" on the unlock characteristic and try writing the values to it. Testing on windows is not super reliable, as you cannot tell easily if the problem is just some windows pairing issue or you doing something wrong in the communication commands.

TreeviewGera, 2025-08-19:
I've tried it. If I do it over and over on nrfconnect, eventually I do get a response and the first square appears on the device, replacing the P. But nothing else happens. And I haven't been able to replicate that with the python script for some reason...

I've now pivoted to try reverse engineering the official .apk. I believe I identified a function that constructs the byte array in question:

```java
public static synchronized byte[] e(byte[] bArr) {
    DebugLog.eLogKind elogkind = DebugLog.eLogKind.S;
    if (bArr != null && bArr.length == 4) {
        byte[] bArr2 = new byte[20];
        bArr2[0] = 17;
        System.arraycopy(bArr, 0, bArr2, 1, bArr.length);
        DebugLog.eLogKind elogkind2 = DebugLog.eLogKind.E;
        return bArr2;
    }
    DebugLog.r("[OHQ]", "OHQSysCommand", "createTransactionCommand", DebugLog.eLogKind.M, "illegal argument");
    return null;
}
```

It seems you were correct, there is a 4 byte key, and the 0x11 (17 in decimal) is prepended to it. The final array is defined to be 20 bytes long as well as you can see. The 4 byte key is randomly generated earlier with java's `SecureRandom()`.

Anyway I was gonna investigate more but something came up so I'll have to pause this for now. I'll leave you with a file I found that may help with info on this and all other models as well:

[Memory_Map.zip](https://github.com/user-attachments/files/21845855/Memory_Map.zip)

userx14, 2025-08-19:
> If I do it over and over on nrfconnect, eventually I do get a response and the first square appears on the device, replacing the P. But nothing else happens.

That's as expected. After writing to the unlock characteristic it waits for the rest of the commands on the tx and rx channel to read the actual data, the rotating squares start when the start transmission message `0800000000100018` is sent to the rx channel. The device freezing with one square when not recieving the follow up communication also happens to the other omron devices. 

I don't get why it is random if it works or not. Did you just write to the unlock characteristic and always the same code? 
Could it be some timing thing? I mean the original code reads some manufacturer strings like device name and version number which takes some time.

TreeviewGera, 2025-08-20:
Yeah the random thing is weird. I just keep writing to the unlock characteristic over and over, switching around between different codes. Eventually one of them will produce a response, usually after 5-10 writes. It's quite weird but yeah. I also thought it could be a timing thing but haven't been able to identify any pattern so far. Good to know the square thing is expected.

On the script if I remove the while loops and associated checks, I do eventually see the first square appear. With a few more changes some subsequent squares appear as well, but eventually the script crashes. I'm not sure if that's progress or not. The main issue seems to be that the device just refuses to send reply notifications for some reason...

Would it be possible to get data relying only on direct reads rather than notifications?

TreeviewGera, 2025-08-20:
Yeah yeah, I actually fixed that by commenting the extra TX and RX UUIDs in the respective arrays, and changing this line like:
```
    deviceDataRxChannelIntHandles = [31] #[0x360, 0x370, 0x380, 0x390]
```

The issue now is not freezing, it's an error. The problem is `self.rxDataBytes` is never set because no callback ever runs. So then this line:
```
            eepromBytesData += await self._readBlockEeprom(startAddress, nextSubblockSize)
```
tries to add some data to None.

Of course I had to do a lot of code commenting, basically ignoring all checks, to even get to that point in the first place...

userx14, 2025-08-20:
Could you fork the repository on your account and upload the changed python file there, just so I have a reference what code you are currently using?

Did you check with nrfconnect if you get a notify callback when you are subscribed to the rx channel and send `0800000000100018` to the tx channel?

I mean yes you can get the value of the bt characteristic with direct reads instead of the callback, but by the bluetooth spec you should get a callback when the data changes, so it should not help.

userx14, 2025-08-20:
Here would be the proposed changes to the callback, not sure if I got everything:
https://github.com/userx14/omblepy/tree/hem7144t2
Please let me know if this one works, I can't really test without a device that has this behavior.

TreeviewGera, 2025-08-20:
Nice! I was able to use that to get data from the device with your changes. I just had to comment this stuff out because the device still refuses to reply there:
https://github.com/TreeviewGera/omblepy/blob/hem-7144t2/omblepy.py#L186-L191

And also made sure to define these to avoid an error later on:
https://github.com/TreeviewGera/omblepy/blob/hem-7144t2/deviceSpecific/hem-7144t2.py#L25-L26

This now works and writes some values to the csv file. The issue now is the record format is still not quite correct. SYS/DIA/BPM values are good, as is the year. But the rest of the datetime values are wrong and cause some entries to get skipped. Here's a breakdown of the data I have:

```
0       8       16      24      32      40      48      56      64      72      80      88      96      104

62      4c      5e      19      96      21      d4      1a      80      00      01      00      4b      b4
50      45      3f      19      ae      61      c9      1a      80      20      02      00      81      7e
49      47      3e      19      ae      21      2a      1b      80      00      03      00      7e      81
5b      47      45      19      ae      61      1f      1c      80      40      04      00      0e      f1
52      45      40      19      b1      21      25      18      80      01      05      00      85      7a
5e      47      38      19      f2      21      20      1a      80      00      06      00      c9      36
54      47      36      19      8e      22      0a      1e      80      10      07      00      59      a6

0110001001001100010111100001100110010110001000011101010000011010100000000000000000000001000000000100101110110100
0101000001000101001111110001100110101110011000011100100100011010100000000010000000000010000000001000000101111110
0100100101000111001111100001100110101110001000010010101000011011100000000000000000000011000000000111111010000001
0101101101000111010001010001100110101110011000010001111100011100100000000100000000000100000000000000111011110001
0101001001000101010000000001100110110001001000010010010100011000100000000000000100000101000000001000010101111010
0101111001000111001110000001100111110010001000010010000000011010100000000000000000000110000000001100100100110110
0101010001000111001101100001100110001110001000100000101000011110100000000001000000000111000000000101100110100110

DATE		TIME	SYS	DIA	BPM
12/08/2025	22:43:  123	76	94
13/08/2025	14:43:  105	69	63 (heart icon)
13/08/2025	14:44:  98	71	62
13/08/2025	14:48:  116	71	69 (heart icon)
13/08/2025	17:32:  107	69	64
15/08/2025	18:40:  119	71	56
20/08/2025	14:56:  109	71	54
```

Top is the raw records in hex, middle is the same but in binary, and bottom is the values I got off the app. The app won't give me seconds. The "heart icon" is the irregular heartbeat mark. Times are in my local timezone, to convert to UTC you'd have to add three hours.

I tried to work out the offsets but I can't really find a configuration that properly maps all values... Stuff ends up overlapping and such. Any ideas?

TreeviewGera, 2025-08-20:
Yeah I see, that makes sense. And it's consistent with the `VitalDataIndexes.json` file. E.g.:
```json
            "minute":{
                "addressOffset":6,
                "size":2,
                "startBit":6,
                "bitSize":6,
                "signed":0
            },
            "second":{
                "addressOffset":6,
                "size":2,
                "startBit":0,
                "bitSize":6,
                "signed":0
            }
```

It can be read out of the byte array like:
```python
def extract(data, offset, size, start_bit, bit_size):
    v = int.from_bytes(data[offset:offset + size], byteorder='little')
    v >>= start_bit
    v &= (2 ** bit_size - 1)
    return v
```

And that confirms you got the seconds right too. The json files defines a whole bunch of other values too but has no description of what they are so it's unclear.

Anyway thank you for all your help, I consider this solved now.

userx14, 2025-08-20:
Now that I look at it again, I would say that the device is actually deviceEndianess = "little". 
This also explains why the minute data is so fractured.